tesseract-robotics/tesseract #19: kdl_env seems to incorrectly load the ACM from SRDF

arocchi, 2018-08-29:
In the implementation of [KDLEnv, at line 150](https://github.com/ros-industrial-consortium/tesseract/blob/3ce8c8ba252318ee02155401ef1d3d17eed12bc4/tesseract_ros/src/kdl/kdl_env.cpp#L150), SRDF's ACM is loaded in a way such that disabled collisions in the SRDF (only) are added as allowed in KDLEnv (basically inverting the ACM).

```
for (const auto& pair : srdf_model_->getDisabledCollisionPairs())
{
  allowed_collision_matrix_->addAllowedCollision(pair.link1_, pair.link2_, pair.reason_);
}
```

Levi-Armstrong, 2018-08-30:
The allowed collision matrix in this case only stores if two objects are allowed to be in collision. The naming is a little confusing.

arocchi, 2018-08-30:
@Levi-Armstrong thank you, I think I misunderstood. so it's the same as calling `collision_detection::AllowedCollisionMatrix::setEntry(name1, name2, true)` in moveit, so that if the collision pair is  "disabled" it means that the collision is "allowed". I am getting weird results in TrajOpt (from `trajopt_ros`) as it looks like self-collisions are not being checked, so I thought I found the culprit here but - I was ultimately victim of the double negation in collision APIs :+1: 
Thanks for clarifying.

closed 2018-08-30 by arocchi