tesseract-robotics/tesseract #24: Duplicates in the list of collisions

AndyZe, 2018-09-08:
Working with collision_monitoring from commit 683f684, it looks like every collision entry is mirrored in the /tesseract_monitoring_contacts/contact_results. Is that desirable? I think once would be enough.

```
constacts: 
  - 
    distance: 0.0769603028893
    type_id: [0, 0]
    link_names: [link_l, link_r]
.
.
.
    distance: 0.0769603028893
    type_id: [0, 0]
    link_names: [link_r, link_l]
```

Levi-Armstrong, 2018-09-08:
This should be fixed in the latest version but I need to update the contact monitor. I will take a look and update the monitor. I let you know when it has been updated.

AndyZe, 2018-09-08:
Cool. Well I was working on a patch right now, but feel free to close my PR if yours works.

Levi-Armstrong, 2018-09-10:
I have upgraded the contact monitor in PR #26 and it appears the latest changes have fixed it, but I have done limited testing so you may want to confirm.

AndyZe, 2018-09-12:
Unfortunately I ran into this error while launching the contact monitor. I'm using a Gazebo SIA5 simulation which seems to be fine. I can plan/execute with MoveIt! and echo /joint_states.

> tesseract_monitoring_contacts_node: /home/me/Desktop/catkin_ws/src/tesseract/tesseract_ros/src/kdl/kdl_joint_kin.cpp:297: bool tesseract::tesseract_ros::KDLJointKin::init(urdf::ModelInterfaceConstSharedPtr, const std::vector<std::__cxx11::basic_string<char> >&, std::__cxx11::string): Assertion `jnt.getType() != KDL::Joint::None' failed.

Levi-Armstrong, 2018-09-13:
This assert is triggered because the the kdl joint tree was passed a set of joints that are adjustable but one of the joints past is not an adjustable joint it is of type none.

closed 2019-10-18 by Levi-Armstrong