tesseract-robotics/tesseract #27: "Failed to calculate jacobian" when using KDLJointKin

arocchi, 2018-09-10:
it looks to me like `KDLJointKin` does not implement successfully `addAttachedLink` (`parent_link_name` is never stored) and thus also `calcJacobian` (if we request the Jacobian of an attached link it will fail instead of computing the Jacobian of its parent link and then transforming it into the attached link).

The problem became apparent when using TrajOpt with a manipulator with an attached object.

Levi-Armstrong, 2018-09-11:
It looks like I did not update this to handle attached links like what is done in the chain library. This should not be to hard to update.

Levi-Armstrong, 2018-09-21:
I should have PR posted today with a fix I am currently writing some unit tests.

Levi-Armstrong, 2018-09-24:
@arocchi I believe this issue has been resolved with PR #32 can you confirm?

arocchi, 2018-10-02:
confirmed, thanks Levi.

closed 2018-10-02 by arocchi