tesseract-robotics/tesseract #43: ROS 2 Support

Levi-Armstrong, 2018-10-31:


Levi-Armstrong, 2018-10-31:
@gavanderhoorn I am interested in supporting both ROS 1 and ROS 2 but not sure what is the best/preferred approach. Is it possible to have single branch support both ROS 1 and ROS 2? Also interested in your thoughts about how best to accomplish this.

Levi-Armstrong, 2018-12-17:
@gavanderhoorn Related to the question above. Several of the package here don't necessarily require ros and could easily be pure cmake packages. In anticipation for adding ROS 2 support is there any reason not to make them pure cmake packages and include a package.xml that allows for them to be build in a ros workspace?

marip8, 2019-02-05:
@Levi-Armstrong have you considered using [`console_bridge`](http://wiki.ros.org/console_bridge)? I'm not sure how it will be supported in the future or how it will interface with logging in ROS2 but I think its at least an existing solution for ROS-style logging in ROS-independent packages

gavanderhoorn, 2019-02-05:
@marip8: I've also suggest `console_bridge` to @Levi-Armstrong. It's a decent suggestion: it has been packaged for ROS 2 usage apparently ([ros2/console_bridge_vendor](https://github.com/ros2/console_bridge_vendor)).

Levi-Armstrong, 2019-02-05:
Looks like a good solution.

Levi-Armstrong, 2019-02-06:
@gavanderhoorn I must have missed it when you pointed out this as a solution.

gavanderhoorn, 2019-02-06:
I sent you an email about it on 2019-01-30.

Levi-Armstrong, 2019-02-09:
@gavanderhoorn I found the email and thank you.

Levi-Armstrong, 2019-02-13:
@gavanderhoorn So I have finished isolating the tesseract_collision in this [branch](https://github.com/Levi-Armstrong/tesseract-1/tree/feature/ContactShapes/tesseract_collision). It currently the only catkin depends is bullet3_ros but this will eventually go away. I am open to solution on how to best release this to support both ROS1 and ROS2.

mpowelson, 2020-03-18:
@Levi-Armstrong Can this now be closed? [Tesseract_ros2](https://github.com/ros-industrial-consortium/tesseract_ros2)

closed 2020-03-19 by Levi-Armstrong