tesseract-robotics/tesseract #62: FreeCAD integration

johnwason, 2018-12-28:
FreeCAD has a (seemingly very limited) robot workbench module that can be used to set waypoints and specify paths.

https://www.freecadweb.org/wiki/Robot_Workbench

Having a tool like this integrated with Tesseract would provide capabilities similar to commercial robot planning software.

Levi-Armstrong, 2018-12-28:
I have been very interested in building this. I have used freecad before and developed a simple plugin before.  I have been discussing this internally whether to integrate into freecad or something custom. I personally vote for freecad myself. I think it would be good to discuss at the next developers meeting.

jf---, 2019-07-04:
perhaps interface with `tesseract` via `roslibpy`. If the drawing classes are abstract, than it becomes easy to adapt to your CAD package of preference, FreeCAD, pythonocc, rhino come to mind

jf---, 2019-12-10:
getting back to this issue - what's pretty amazing is that a great way to install `freecad` is via [conda](https://github.com/FreeCAD/FreeCAD_Conda). having such a conda recipe might be a nice way to build towards this goal? These days `ROS` also comes with pre-build conda packages, hence it might be a good idea IMO.

Currently, how to build the project and which platforms are supported is somewhat underdocumented and its easy to separate the conda specifics from general build mechanics

Levi-Armstrong, 2020-09-04:
We plan to leverage Ignition Robotics software packages.

closed 2020-09-04 by Levi-Armstrong