tesseract-robotics/tesseract #66: Tesseract Development TODO List

mpowelson, 2019-01-28:
- [ ] Continuous collision checking between two moving objects
- [ ] Upgrade data structure to support graphs
- [ ] Upgrade to support ROS2 #43 
- [ ] Upgrade to use Bullet3 
- [ ] TrajOpt Hybrid planner to seed trajectory

mpowelson, 2019-11-01:
I believe the ROS 2 wrapper is underway. @schornakj  could you link to it here?

schornakj, 2019-11-13:
@mpowelson The ROS2 versions of the `tesseract_ros` packages will live here: [https://github.com/ros-industrial-consortium/tesseract_ros2](https://github.com/ros-industrial-consortium/tesseract_ros2)

The main repo is still waiting on some PRs to be merged. In the meantime, my latest dev branch is here: [https://github.com/schornakj/tesseract_ros2/tree/feature/migrate-tesseract-monitoring]( https://github.com/schornakj/tesseract_ros2/tree/feature/migrate-tesseract-monitoring)

mpowelson, 2020-10-20:
@Levi-Armstrong Is anything here still relevant or should we close?

Levi-Armstrong, 2020-10-21:
I would just move the sub graph items to the new one and we can mark themes nice to have. We already support adding scene graphs to another but the are not modeled as a sub graph that boost does support.

mpowelson, 2020-10-21:
All issues have been addressed with the exception of #420.

closed 2020-10-21 by mpowelson