tesseract-robotics/tesseract #68: RVIZ crashes when trying to add tesseract_trajectory or when trying to load robotmodel with 'base_link'

lakshmip001, 2019-02-11:
I get the following error when trying to load trajectory or robot model

terminate called after throwing an instance of 'Ogre::ItemIdentityException'
  what():  OGRE EXCEPTION(4:ItemIdentityException): An object of type 'Entity' with name 'Robot Link4' already exists. in SceneManager::createMovableObject at /build/ogre-1.9-mqY1wq/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreSceneManager.cpp (line 6832)
Aborted (core dumped)

Kindly please help me how to overcome this error and why do I get this?

Levi-Armstrong, 2019-02-15:
This is a oger shape naming conflict. I need to figure the best way to solve the issue.

Levi-Armstrong, 2019-07-14:
This is resolved in the master branch.

closed 2019-07-14 by Levi-Armstrong