tesseract-robotics/tesseract #70: Potential release - and issue with bullet3_ros with other packages

wxmerkt, 2019-02-12:
Hi guys!
Thank you very much for the great work in putting together this framework! I am exploring `tesseract_collision` and really like the abstraction :-).

However, using the package leads to conflicts in our catkin workspace - particular due to the following line:
https://github.com/ros-industrial-consortium/tesseract/blob/4c6d86028bbc08c39606acd5c1b1ba79d69874ce/tesseract_ext/bullet3_ros/CMakeLists.txt#L15
which breaks some of our buildspace logic (and wouldn't work e.g. on the buildfarm).

As thus my questions/request:
- Are there additional bug fixes in upstream bullet that are not yet in the system packages `libbullet-dev`? [I guess 'yes']
- Is there a way to get `bullet3_ros` or `tesseract` as a whole released upstream via the ROS buildfarm? I know there is likely little appetite for a newer version of a system-provided library (and have personally gone through that process a few times - with [`fcl_catkin`](https://github.com/wxmerkt/fcl_catkin) and [`pybind11_catkin`](https://github.com/ipab-slmc/pybind11_catkin)). If there is clear benefit and no downside in library conflicts, it may be worth trying.
- E.g., the `fcl_catkin` we release upstream (as of today) contains @Levi-Armstrong's Convex constructor fix. However, we did not package `libccd` (and probably should).
- If not, can the logic in `bullet3_ros` be updated not to create the folder on a devel-only workspace?

Thank you very much,
Wolfgang

Levi-Armstrong, 2019-02-13:
>   Are there additional bug fixes in upstream bullet that are not yet in the system packages `libbullet-dev`? [I guess 'yes']

You are correct the current version at least on Kinetic is from 2015 and they have made a lot of improvements.

> Is there a way to get `bullet3_ros` or `tesseract` as a whole released upstream via the ROS buildfarm? I know there is likely little appetite for a newer version of a system-provided library (and [...]

I do plan on eventually releasing tesseract but will need to look how to get bullet3 released. This may not be an issue on melodic but not sure.

>     E.g., the `fcl_catkin` we release upstream (as of today) contains @Levi-Armstrong's Convex constructor fix. However, we did not package `libccd` (and probably should).

That is great to hear. For libccd i remeber reading some where that it stated to include libccd from source because of bug fixes that had not been released but not sure if that was in bullet or fcl. 

 > If not, can the logic in `bullet3_ros` be updated not to create the folder on a devel-only workspace?

@marip8 I believe you made this change, would there be any issue with changing this to devel?

Levi-Armstrong, 2019-02-13:
@wxmerkt I looked at `fcl_catkin` but may need some help. I noticed that you made fcl a catkin package, are not able to release it as a pure cmake leveraging the package.xml? Is this a limitation of the build farm? I ask because I have no experience releasing a non catkin package through the ros build farm.

marip8, 2019-02-13:
> If not, can the logic in `bullet3_ros` be updated not to create the folder on a devel-only workspace?

@Levi-Armstrong I was not involved in this change. It looks like it came from [this commit](https://github.com/ros-industrial-consortium/tesseract/commit/408ce8474f7cd57f90257aac65cfc1f376445d98)

wxmerkt, 2019-02-13:
Thank you for the detailed response!

We decided to make FCL a catkin package and namespace it such that it does not conflict with system-installed FCL, i.e., when including we run:
```cmake
find_package(catkin REQUIRED COMPONENTS fcl_catkin)
include_directories(${catkin_INCLUDE_DIRS}/fcl_catkin)
```
as the headers are in `fcl_catkin/fcl` (we had to make sure there can't be any conflicts). It may have been possible to build as a CMake project as well and I am happy for someone with more catkin-expertise than I do to have a stab at that :).

If it works for you (CMakeLists have to change slightly), I could assist with changing `bullet3_ros` in a similar way as we do for `fcl_catkin` and trying to get it released via the buildfarm. There clearly is benefit (but we may have to do some convincing...).

arocchi, 2019-02-13:
On a maybe related note: probably having a namespaced fcl like this could be also a good idea for `tesseract`. As of now, I am personally maintaining a version of `tesseract` [without fcl](https://github.com/arocchi/tesseract/tree/only_bullet) for our internal use as it conflicts with our system-level fcl.

wxmerkt, 2019-02-14:
@arocchi I am doing the same here and using two different workspaces (one install, one devel that extends on the install). We dont have bullet conflicts but that's likely because we aren't using another library that depends on it. Note, we have only namespaced the header location and made sure it doesn't get linked to the wrong FCL - users may still end up loading code as plug-ins from two different versions.

Levi-Armstrong, 2019-02-15:
> On a maybe related note: probably having a namespaced fcl like this could be also a good idea for `tesseract`. As of now, I am personally maintaining a version of `tesseract` [without [...]

So how do I go about namespace fcl? I something special I can add to the CMakeLists.txt to accomplish this?

Levi-Armstrong, 2019-07-19:
@wxmerkt So how do I go about namespace fcl? Is there something special I can add to the CMakeLists.txt to accomplish this?

wxmerkt, 2019-07-20:
Hi @Levi-Armstrong,
I am not sure - there is no flag/option I know of in the CMake to accomplish this (perhaps private include and linking?). I think you would have to manually set the namespace or, what we do and it is a workaround/hack, install into a subdirectory and then use a CMake hook to set the correct paths (but with the same namespace). Please let me know if you find a good solution - would love to learn :-)

Best,
Wolfgang

Levi-Armstrong, 2020-12-24:
@wxmerkt and @arocchi We are about start going through the release process of Tesseract, Tesseract ROS, and TrajOpt and the first thing will be to figure out how to handle Bullet and FCL. The current software can be build with the OS version of Bullet now but the current released version is not built with double precision and for FCL it requires version 6.0 which is not released on ubuntu 20.04 yet and not sure if it will be. Have either found a good way to release these?

Levi-Armstrong, 2020-12-24:
@wxmerkt Should I be able to use your `ros-noetic-fcl-catkin` and could I assume I could do the same as you have for bullet? Create a bullet_catkin package and release it.

wxmerkt, 2020-12-25:
Hi @Levi-Armstrong,
For Noetic you can go directly with `ros-noetic-fcl` - this package is directly from upstream and got released after fcl_catkin once it was discovered that the Noetic-supported OS versions use different libfcl-dev package versions. The 'tricks' we used for fcl_catkin (isolation of headers in a sub-directory and CFG_EXTRAS to amend paths) aren't the best way to do things, but were required pre-Noetic to get the latest version into the ROS buildfarm without causing conflicts with libfcl-dev. I am not sure whether the same should be used to vendor bullet or not (or whether there are better ways to do this!), I am afraid.

Levi-Armstrong, 2020-12-28:
@wxmerkt Thank you for the feedback. I was thinking of statically linking against bullet for tesseract_collision because I would like to use bullet built with double precision. Do you know if I will have issues using cmake `ExternalProject_Add` during the release process to build the static bullet libraries?

wxmerkt, 2020-12-28:
If you specify all required `build_dependency` (incl. git if not using a zip/tar ball) and use writable directories (`PROJECT_BINARY_DIR`), it should be fine I reckon? We use `ExternalProject_Add` for `fcl_catkin` and it works.

Levi-Armstrong, 2020-12-29:
@wxmerkt Thank you.

Levi-Armstrong, 2021-01-09:
It looks like fcl 6.0 is released on ros noetic build farm `ros-noetic-fcl` and the Ubuntu team now release bullet for both float32 and float64. The method for using float64 version over the float32 is shown below which should resolve this issues at least for noetic.

```
find_package(Bullet REQUIRED CONFIGS BulletConfig-float64.cmake BulletConfig.cmake)
IF(NOT ${BULLET_DEFINITIONS} MATCHES ".*-DBT_USE_DOUBLE_PRECISION.*")
  message(WARNING "Bullet does not appear to be built with double precision, current definitions: ${BULLET_DEFINITIONS}")
endif()
```

closed 2021-01-09 by Levi-Armstrong