tesseract-robotics/tesseract #71: Impact of KinematicFilter assert

wxmerkt, 2019-02-12:
Hi,
I am creating a scene based on the `tesseract_collision` package in which I load robot links (with mask id `BodyType::ROBOT_LINK`) and environment objects (with mask id `BodyType::ROBOT_ATTACHED`).

When compiling in debug mode I frequently receive the following assert:
https://github.com/ros-industrial-consortium/tesseract/blob/8379d427cda2e413216031b9142335aa75a0a71e/tesseract_collision/include/tesseract_collision/bullet/bullet_utils.h#L420-L421
(asserts are removed in Release mode)

I'd like to understand this better to see where I am going wrong - @Levi-Armstrong do you have any insights on the origin of/need for this check here, and how it is for instance set for `trajopt_ros`?

Thank you very much for the insights,
Wolfgang

mpowelson, 2019-02-12:
@Levi-Armstrong For what it's worth, the car_seat_demo has a similar problem. I'm not sure if it's related, but it throws an error here.

https://github.com/ros-industrial-consortium/tesseract/blob/8379d427cda2e413216031b9142335aa75a0a71e/tesseract_collision/src/bullet/bullet_cast_bvh_manager.cpp#L211

wxmerkt, 2019-02-12:
Unrelated to the original issue: I couldn't get any of the demos to visualise :'( [they solve correctly but the tesseract_rviz plugin does not show anything - I have all the dependencies]

mpowelson, 2019-02-12:
If you add a TesseractTrajectory, you should be able to see it moving. I sometimes have to click the topic selected a couple of times to get it to show up. I'm not sure why.

Levi-Armstrong, 2019-02-13:
> I'd like to understand this better to see where I am going wrong - @Levi-Armstrong do you have any insights on the origin of/need for this check here, and how it is for instance set for `trajopt_ros`?

The line of code you highlighted is to check for the case when performing continuous collision checking if both objects are moving. It currently only supports checking moving objects to static objects for continuous collision checking. It is possible, but I need to workout the math to calculate the nearest points and contact time two moving objects. 

Do you have an example I can look at to determine what is going on?

wxmerkt, 2019-02-13:
Thank you for the response and explanation - that makes sense and mirrors my understanding from the paper. 
In my case, I only check one object A (moving) against a static object B at a time and disable all other active collision objects:
```cpp
for (const auto& obj : bullet_cast_bvh_manager_->getActiveCollisionObjects())
        bullet_cast_bvh_manager_->disableCollisionObject(obj);

bullet_cast_bvh_manager_->enableCollisionObject(A);
bullet_cast_bvh_manager_->enableCollisionObject(B);
```

I then set the transforms:
```cpp
bullet_cast_bvh_manager_->setCollisionObjectsTransform(A, T_A_beg, T_A_end);
bullet_cast_bvh_manager_->setCollisionObjectsTransform(B, T_B_beg);
```

While this should resolve the `assert` issue (I will report back if it re-appears - it happened on a static object [but I may have set the beginning and end transform as the same]), my question is - are you casting all collision links associated with the manipulator at the same time against _each single_ environment object, i.e., 

(a) All robot links cast as moving vs 1 static object (for each object...)
(b) Each individual link cast as moving vs each individual world object.

Thank you very much for your comments and insights :)

Levi-Armstrong, 2019-02-14:
Lets see. So currently because you have the ability to set which links are active, for every collision shape (link can contain multiple collision shapes) i store a cast and static representation of the collision shape.  Then when you set/change the active links I remove/add objects to the broadphase. When you set the transforms for moving objects I update the bounding box in the broadphase. The way bullet works is it will check every collision shape in the broadphase against the broadphase. It leverage the filters to quickly discard unnecessary checks.

wxmerkt, 2019-02-17:
Thank you for the detailed explanation. Do I need to specifically select the cast/static representation or how does it activate the _correct_ one? I.e., I am still triggering the KinematicFilter assert when setting the following:
```
bullet_cast_bvh_manager_->setCollisionObjectsTransform(o1, T1_beg, T1_end);

// TODO: Note, the cast method currently does not support both objects moving
// - thus, we are only updating the second object's transform with T2_beg here.
bullet_cast_bvh_manager_->setCollisionObjectsTransform(o2, T2_beg); //, T2_end);
```

Do I need to set the `mask_id` on object creation somehow?

Levi-Armstrong, 2019-02-18:
Have you already set the active collision object, prior to setting the transforms? After adding the collision objects you identify which objects can move by calling the setActiveCollisionObjects function.

wxmerkt, 2019-02-18:
> Have you already set the active collision object, prior to setting the transforms?

Yes, using `enableCollisionObject`. I have now changed it to do `setActiveCollisionObjects` after but it does not detect continuous collisions anymore.

This is my code:
```cpp
// Disable active collision objects
for (const auto& obj : bullet_cast_bvh_manager_->getActiveCollisionObjects())
    bullet_cast_bvh_manager_->disableCollisionObject(obj);

// NB: This is what I had previously and worked, but took very, very long.
// bullet_cast_bvh_manager_->enableCollisionObject(o1);
// bullet_cast_bvh_manager_->enableCollisionObject(o2);

// Update object to cast only if there is motion between beginning and end (otherwise segfault)
Eigen::Isometry3d T1_beg = tesseract_convert::KDL2Isometry3d(tf1_beg);
Eigen::Isometry3d T1_end = tesseract_convert::KDL2Isometry3d(tf1_end);
Eigen::Isometry3d T2_beg = tesseract_convert::KDL2Isometry3d(tf2_beg);

// Set collision object transforms - the following check is required as
// Bullet segfaults in AABB creation if there is no motion.
if (T1_beg.isApprox(T1_end))
{
    bullet_cast_bvh_manager_->setCollisionObjectsTransform(o1, T1_beg);
}
else
{
    bullet_cast_bvh_manager_->setCollisionObjectsTransform(o1, T1_beg, T1_end);
}
bullet_cast_bvh_manager_->setCollisionObjectsTransform(o2, T2_beg);

// Set active collision objects
bullet_cast_bvh_manager_->setActiveCollisionObjects({ o1 });

// Contact test...
```

From an understanding point of view, is it correct to (here `o1` is the object to be cast, `o2` is static world object):
1. Disable all active collision objects.
2. Set transform for o1 - using either one (if no motion) or two transforms.
3. Set static transform for o2
4. Set active collision objects o1, o2
5. Contact test

Levi-Armstrong, 2019-02-18:
I notice that you start be disabling all current active objects does do objects o1 and o2 enabled? Based on what is shown above that either o1 or o2 is disabled. I see you note about calling `enableCollisionObject` taking a very long time but it should all it is doing is setting a bool flag. It does call find for a std::map but I would not expect it to take very long.

Recommended procedure:

- Enable all objects that you want to be considered during collision checking
- Set the active collision objects
- Set both active and static collision objects transforms
- Perform contact test.

wxmerkt, 2019-02-20:
Thank you, this helped a lot.

closed 2019-02-20 by wxmerkt