tesseract-robotics/tesseract #73: CartVelTermInfo() parameters adjustment to acheive trapezoidal Cartesian Velocity

lakshmip001, 2019-02-15:
Time of start is generated by trajopt but when the trajectory is given to IPTP it gives new time of start and the velocity is profile is as shown in the below figure:

[image]

And I am able to define constraint on CartVelTermInfo() as shown below but not for each waypoint

std::shared_ptr<CartVelTermInfo> cv(new CartVelTermInfo);
      cv->term_type = TT_COST;
      cv->first_step = 0;
      cv->last_step = pci.basic_info.n_steps-1;
      cv->max_displacement = 0.05;
      cv->link = "tool0";

Kindly please help me to understand how to play with the parameters to acheive trapezoidal velocity profile or approximated velocity profile.

mpowelson, 2019-02-15:
I believe this should be a TrajOpt issue?

lakshmip001, 2019-02-15:
Yes, I am not sure what parameters indicate. So , knowing them will be more helpful in understanding parameters.

mpowelson, 2019-02-15:
In general, you are not going to be able to use Trajopt/Tesseract at the moment without reading the docs in the code directly. Checkout the header for that function. It should be in problem_description.

lakshmip001, 2019-02-15:
I tried searching through the header file , but didn't find explaination what parameters represent .

mpowelson, 2019-02-15:
`/**
 \brief Applies cost/constraint to the cartesian velocity of a link

 Constrains the change in position of the link in each timestep to be less than
 max_displacement
 */
struct CartVelTermInfo : public TermInfo
{
  /** @brief Timesteps over which to apply term */
  int first_step, last_step;
  /** @brief Link to which the term is applied */
  std::string link;
  double max_displacement;
  /** @brief Used to add term to pci from json */
  void fromJson(ProblemConstructionInfo& pci, const Json::Value& v) override;
  /** @brief Converts term info into cost/constraint and adds it to trajopt problem */
  void hatch(TrajOptProb& prob) override;
  DEFINE_CREATE(CartVelTermInfo)

  /** @brief Initialize term with it's supported types */
  CartVelTermInfo() : TermInfo(TT_COST | TT_CNT) {}
};`

I'd have to look a the code to give you much more than that. I did not write this one.

closed 2019-08-01 by mpowelson