tesseract-robotics/tesseract #87: add python interface?

jacknlliu, 2019-06-18:


Levi-Armstrong, 2019-06-19:
This should be available soon.

Levi-Armstrong, 2019-07-02:
@KevinWarburton

jf---, 2019-07-04:
FWIW, if refactoring is to be done, `pybind11` is quite a bit more convienent than generating bindings via boost

Levi-Armstrong, 2019-07-04:
We have been using SWIG.

jf---, 2019-07-05:
Cool thing with `pybind11` is that it supports modern C++, where SWIG is pretty much in maintenance mode

traversaro, 2019-07-26:
> Cool thing with `pybind11` is that it supports modern C++, where SWIG is pretty much in maintenance mode. Stoked that bindings are in the pipline

Which specific features of "modern" C++ are you referring to? SWIG has some support for features of C++11 ( http://www.swig.org/Doc4.0/CPlusPlus11.html ), C++14 ( http://www.swig.org/Doc4.0/CPlusPlus14.html ) and C++17 (http://www.swig.org/Doc4.0/CPlusPlus17.html), but perhaps there are specific features only covered by `pybind11` .

johnwason, 2019-11-27:
The Python bindings should be ready. 

https://github.com/johnwason/tesseract_python

I still need to develop unit tests and examples.

jf---, 2019-11-28:
That's duly impressive @johnwason... 
Given the `ROS` depencies, and [conda](https://medium.com/@wolfv/ros-on-conda-forge-dca6827ac4b6) recently being able to install `ROS`, is making such a `conda` package an interesting idea?

johnwason, 2019-11-28:
I broke up the Python into two module, `tesseract` and `tesseract_ros`. If you don't use the `tesseract_ros` package, there aren't any ROS dependencies.

johnwason, 2019-12-09:
I have pushed some bug fixes and tests. I could not get the industrial_ci to work though. I need to build SWIG from source, but it is being built outside of the container and is not available after the industrial_ci script starts. Any suggestions on how to fix this?

@Levi-Armstrong do you want to merge the Python wrappers into the main repo?

Levi-Armstrong, 2019-12-10:
Will a custom build of SWIG be required going forward? If so you can add swig to tesseract_ext possible using cmakes [ExternalProject_Add](https://github.com/ros-industrial-consortium/tesseract_ext). Each of the packages currently use this feature for building external dependencies. Another option would be to create a docker with your version installed. I most likely will do this for the packages in tesseract_ext to speed up CI builds.

> do you want to merge the Python wrappers into the main repo?
Absolutely

johnwason, 2019-12-10:
For externalproject_add, we should probably check to see if the correct version is available before downloading it, and also have a flag to disable the behavior. It may prove to be a problem for other build systems like vcpgk.

Levi-Armstrong, 2019-12-10:
Yea my next step is to take what you have done in porting tesseract and trajopt into the main repository.

Levi-Armstrong, 2020-01-13:
@johnwason I wanted to add swig to tesseract external. Did you already create this?

johnwason, 2020-01-13:
No, I have not worked on tesseract_ext.

Levi-Armstrong, 2020-01-13:
Ok thanks.

closed 2020-01-13 by Levi-Armstrong