tesseract-robotics/tesseract #88: Uninitialized reads reported in Valgrind

j-petit, 2019-06-26:
### Error Description
When running Valgrind on the [this unit test](https://github.com/ros-industrial-consortium/tesseract/blob/master/tesseract_ros/test/kdl_env_unit.cpp) I get the following conditional jump on not initialized values messages [valgrind_output](https://github.com/ros-planning/moveit/files/3326052/valgrind_output.txt). The [code snippet in question](https://github.com/ros-industrial-consortium/tesseract/blob/740e6607423e2afbeba04b609e118043c86b7266/tesseract_ros/src/kdl/kdl_env.cpp#L751) deals with reading in the robot links from the URDF and constructing out of them the bullet collision objects. For the first constructed object, this results in an uninitialized read at [this point](https://github.com/ros-industrial-consortium/tesseract/blob/740e6607423e2afbeba04b609e118043c86b7266/tesseract_collision/src/bullet/bullet_utils.cpp#L304). 

### System Description
Description:    Ubuntu 18.04.2 LTS
Release:    18.04
Codename:   bionic

Linux jp-XPS-15-9550 4.15.0-51-generic
x86_64 x86_64 x86_64 GNU/Linux

ROS_DISTRO=melodic
ROS_ETC_DIR=/opt/ros/melodic/etc/ros
ROS_PACKAGE_PATH=/opt/ros/melodic/share
ROS_PYTHON_VERSION=2
ROS_VERSION=1
ROS_ROOT=/opt/ros/melodic/share/ros
ROS_MASTER_URI=http://localhost:11311
ROSLISP_PACKAGE_DIRECTORIES=

Levi-Armstrong, 2019-06-27:
This has been solved in PR #72. We are finalizing some remaining changes this week and should be able to merge next week into master branch.

felixvd, 2019-06-27:
Is there any way you might remember what the problem was or where it was fixed? That PR is 130,000 lines long(!)

Levi-Armstrong, 2019-06-28:
Sorry I do not, We have been rewriting everything so there has been a lot of changes.

mpowelson, 2020-01-08:
@Levi-Armstrong Since #72 has been merged, should this be closed?

Levi-Armstrong, 2020-01-09:
It might be good to rerun valgrind to confirm.

Levi-Armstrong, 2020-09-04:
Closing, please reopen if needed.

closed 2020-09-04 by Levi-Armstrong