tesseract-robotics/tesseract #112: Failed to update rviz environment.

zyjiao4728, 2019-08-21:
I got the following error running car_seat_example in tesseract_ros_examples package:
`[ INFO] [1566358488.704137330]: Failed to update rviz environment`

I also played with the examples and tried to modify the environment. I always get `Failed to update rviz environment` error in `sendRvizChanges()` after removing links or joints, by calling `removeLink()` or `removeJoint()`, from the environment.

We also have some observations which may help you debuging.
1. The `sendRvizChanges()` failed when calling `removeLink()` or `removeJoint()`. Using add/move link is fine.
2. If we set the `rviz` parameter to false in the launch file, then no error prompted and the program can successfully finish.

I am using Ubuntu 18.04 with ROS Melodic and Gurobi optimizer.

Could anyone help me to solve this error? Thanks.

Levi-Armstrong, 2019-08-24:
I will take a look at the demo and see what I can find.

mpowelson, 2019-09-20:
Are you still running into this error?

 @Levi-Armstrong did you ever get a chance to look at this? It runs fine on my 16.04 machine

Levi-Armstrong, 2019-09-22:
Sorry, I have not.

Levi-Armstrong, 2019-12-29:
This is fixed in the latest version.

closed 2019-12-29 by Levi-Armstrong