tesseract-robotics/tesseract #127: Eigen Alignment Issues

mpowelson, 2019-09-20:
TrajOpt and Tesseract have been plagued by Eigen alignment issues for a while (see https://github.com/ros-industrial-consortium/trajopt_ros/issues/27). However it did not seem to be widespread and was limited to only a handful of applications and only on certain people's computers. In working on #117 it suddenly got a lot worse. This was presumably because we switched to using make_shared instead of new when creating pointers, so suddenly none of the EIGEN_MAKE_ALIGNED_OPERATOR_NEW macros worked. Regardless, a class that contains an aligned class must also be aligned, and aligned Eigen types are used in some pretty fundamental classes (e.g Link). Requiring a user to know that they must use new instead of make_shared seems unacceptable.

It turns out that for Eigen 3.2+ this is only an issue on 32 bit machines or when compiling on 64 bit machines with AVX instructions. Codebases like Drake have therefore fixed it by simply disabling AVX instructions in the cmake.

Apparently there is a fix in a newer version of Eigen (could you provide a link? @Levi-Armstrong ). However it is unknown if that will be backported to the system version for still currently common systems like 16.04.

The current plan is to follow Drake's example and disable AVX instructions for most packages in Tesseract.

Levi-Armstrong, 2019-09-20:
Here is the [Eigen PR](http://eigen.tuxfamily.org/bz/show_bug.cgi?id=1409) that has been reference as a fix.

mpowelson, 2019-09-26:
Closing this as this was fixed in #117. If this pops up for anyone in the future comment or reopen this issue, as we may just need to add some more tags to the cmake.

closed 2019-09-26 by mpowelson