tesseract-robotics/tesseract #131: tesseract_urdf parser fails if multiple materials with the same name exist

schornakj, 2019-09-25:
xacro can generate URDFs that contain repeated material definitions. I've included a representative snippet below.

```
...
<link name="wrist_1_link">
    <visual>
      <geometry>
        <mesh filename="package://myrobot_support/meshes/ur10e/visual/wrist1.dae"/>
      </geometry>
      <material name="LightGrey">
        <color rgba="0.7 0.7 0.7 1.0"/>
      </material>
    </visual>
    <collision>
      <geometry>
        <mesh filename="package://myrobot_support/meshes/ur10e/collision/wrist1.stl"/>
      </geometry>
    </collision>
    <inertial>
      <mass value="1.96"/>
      <origin rpy="0 0 0" xyz="0.0 0.0 0.0"/>
      <inertia ixx="0.0051082479567" ixy="0.0" ixz="0.0" iyy="0.0051082479567" iyz="0.0" izz="0.0055125"/>
    </inertial>
  </link>
  <joint name="wrist_2_joint" type="revolute">
    <parent link="wrist_1_link"/>
    <child link="wrist_2_link"/>
    <origin rpy="0.0 0.0 0.0" xyz="0.0 0.135 0.0"/>
    <axis xyz="0 0 1"/>
    <limit effort="54.0" lower="-6.28318530718" upper="6.28318530718" velocity="6.28"/>
    <dynamics damping="0.0" friction="0.0"/>
  </joint>
  <link name="wrist_2_link">
    <visual>
      <geometry>
        <mesh filename="package://myrobot_support/meshes/ur10e/visual/wrist2.dae"/>
      </geometry>
      <material name="LightGrey">
        <color rgba="0.7 0.7 0.7 1.0"/>
      </material>
    </visual>
    <collision>
      <geometry>
        <mesh filename="package://myrobot_support/meshes/ur10e/collision/wrist2.stl"/>
      </geometry>
    </collision>
    <inertial>
      <mass value="1.96"/>
      <origin rpy="0 0 0" xyz="0.0 0.0 0.0"/>
      <inertia ixx="0.0051082479567" ixy="0.0" ixz="0.0" iyy="0.0051082479567" iyz="0.0" izz="0.0055125"/>
    </inertial>
  </link>
  <joint name="wrist_3_joint" type="revolute">
    <parent link="wrist_2_link"/>
    <child link="wrist_3_link"/>
    <origin rpy="0.0 0.0 0.0" xyz="0.0 0.0 0.12"/>
    <axis xyz="0 1 0"/>
    <limit effort="54.0" lower="-6.28318530718" upper="6.28318530718" velocity="6.28"/>
    <dynamics damping="0.0" friction="0.0"/>
  </joint>
  <link name="wrist_3_link">
    <visual>
      <geometry>
        <mesh filename="package://myrobot_support/meshes/ur10e/visual/wrist3.dae"/>
      </geometry>
      <material name="LightGrey">
        <color rgba="0.7 0.7 0.7 1.0"/>
      </material>
    </visual>
    <collision>
      <geometry>
        <mesh filename="package://myrobot_support/meshes/ur10e/collision/wrist3.stl"/>
      </geometry>
    </collision>
    <inertial>
      <mass value="0.202"/>
      <origin rpy="0 0 0" xyz="0.0 0.0 0.0"/>
      <inertia ixx="0.000526462289415" ixy="0.0" ixz="0.0" iyy="0.000526462289415" iyz="0.0" izz="0.000568125"/>
    </inertial>
  </link>
...
```

The stock ROS URDF parser is apparently able to handle this without error, but the new `tesseract_urdf` parser throws the error below. @Levi-Armstrong try running the example workspace I sent you a few days ago.

```
Error:   Error parsing urdf 'link' element for urdf 'ur10e'!
           Error parsing link 'visual' element for link 'shoulder_link'!
             Error parsing visual 'material' element!
               Multiple materials with the same name exist!
         at line 167 in /home/joe/workspaces/descartes_unit_test_ws/src/tesseract/tesseract/tesseract_urdf/src/urdf_parser.cpp
```

I think reasonable behavior for backwards compatibility would be using the first material encountered if duplicates are loaded.

mpowelson, 2019-09-26:
That was intentional. It's amazing how many things the URDF wiki (which appears to be the only document that states what a URDF actually is) leaves up to the design of the parser. However I would agree, if that is what URDF_dom does, then I would say we need to support that.

I think I would probably propose the other way. That re-specifying a material overwrites the previous one. That way if you want to call something LightGrey you don't have to worry about some upstream xacro macro that you didn't write overwriting you.

Levi-Armstrong, 2019-10-11:
This has been addressed

closed 2019-10-11 by Levi-Armstrong