tesseract-robotics/tesseract #132: What's the best way to display the scene in RViz?

haudren, 2019-09-26:
Hello,

I was dusting off some old experiments I made. At that time, I was using `ROSBasicPlotting::plotScene()` to send collision updates to RViz. What's the preferred way nowadays?
- I see that the examples use `sendRvizChanges` under the hood, that makes a service call, but there's a bit of boilerplate associated with revision tracking and so on.
- I also see that there is an `EnvironmentMonitor`, is it usable? Would it allow for similar functionality?

Levi-Armstrong, 2019-10-03:
If it is just for visualization you can just publish the environment information to a topic and then using the TesseractState display set the right topic it should just update. The examples is a good example if you what to ensure that your changes get added to the visualization along with verifying that something has not changed.

haudren, 2019-10-04:
Thanks to you both, I ended up following the examples and using:
- `plotTrajectory` for the trajectory itself
- `getRevision` in conjunction with [`sendRvizChanges`](https://github.com/ros-industrial-consortium/tesseract/blob/d6397ad1e8b2d532581e9c3cf03c6ec5ae2ecf1c/tesseract_ros/tesseract_examples/tesseract_ros_examples/include/tesseract_ros_examples/example.h#L108) to broadcast collision shapes.

closed 2019-10-04 by haudren