tesseract-robotics/tesseract #134: .dae mesh collision geometry loaded from URDF is incorrectly oriented

schornakj, 2019-09-30:
I get different collision results with geometry defined by .dae meshes when using versions of Tesseract before (bef6578) and after (d6397ad) the addition of the new `tesseract_urdf` parser. The URDF looks the same as before when visualized in Rviz, so I think this is because of a difference in how `urdfdom` and `tesseract_urdf` parse the files.

I can demonstrate this by defining the same geometry both as primitive shapes in the URDF and as a multibody .dae object. Assuming that a collision-free path through the waypoints exists, planning will succeed for the primitive-based geometry but may fail for the .dae-based geometry.

My initial guess based on the nature of the failures is that .dae geometry is being rotated by 90 degrees about each object's local Z-axis, though I don't have sufficiently detailed results to conclusively show this.

[I made a repo with some test cases](https://github.com/schornakj/descartes_system_tests). You should be able to replicate this by checking out both versions of Tesseract and running `rosrun descartes_system_tests collision_body_tests` for each. All tests should succeed for bef6578 and the `DescartesMultithreadUnitNearMultibodyDAE` test should fail for d6397ad.

gavanderhoorn, 2019-09-30:
I'm not entirely sure, but you could be running into (something similar to) https://github.com/ros-planning/geometric_shapes/pull/52.

Levi-Armstrong, 2019-09-30:
Oh, the urdf parser was created to support not only meshes but convex_mesh, sdf_mesh, etc. Before this change it was hard coded such that all meshes where considered that they may also be convex meshes because there was no way to distinguish between the two. Now with the new urdf parser you are able to distinguish between the two types, but if you use a mesh type as a collision geometry it currently converts it to a convex hull. I need to remove this hard coded conversion to leave it as a detailed mesh for collision shapes which I missed. If you change your geometry type to a convex_mesh type it should work as it did before. I will also create PR to remove the conversion to detailed convex mesh for detailed meshes.

Levi-Armstrong, 2019-10-01:
Also when using the convex_mesh type, it expects the mesh to only have one polygon face on a given plane, not multiple triangles that make up the face. Unless you know for sure the file type has the mesh encoded correctly, I recommend adding the convert="true" to make sure it is correct.

Levi-Armstrong, 2019-10-01:
PR #135  Should address this issue. I have also added a version number to the tesseract_urdf parser so any urdf missing the version attribute or version less than 2 will be parsed the same as it was prior to the tesseract_urdf.

gavanderhoorn, 2019-10-01:
@gavanderhoorn wrote:
> super pedantic [..]

and so it was ..

schornakj, 2019-10-02:
@Levi-Armstrong I agree that #135 seems to solve this. I'll close this issue once it's merged.

@gavanderhoorn I think (going off sparse details since Levi wrote the fix) the underlying problem was that in the new parser mesh elements were just left unchanged instead of being converted to convex hulls. The generated hull meshes had the correct orientation but the original meshes didn't. This wasn't a problem in the previous mesh parser because it would convert _every_ mesh body to a convex hull. So you're not completely off-base here!

closed 2019-10-03 by Levi-Armstrong