tesseract-robotics/tesseract #142: How to add multiple "robots" to the same Environment?

haudren, 2019-10-08:
I've been experimenting a bit more with Tesseract / TrajOpt and it seems to give good results :+1: 

However I would like to do the following:
- Create an Environment containing my robot via `Tesseract::init(urdf, srdf, locator)`
- Add collision shapes and other objects independently

So far I've been constructing them by hand (create Link, create Joint, then call `addLink`), but I would ideally load them from a URDF file. However a quick look at the documentation tells me that:
- `parseURDF` either returns a `scene_graph` or appends to a `scene_graph`
- `Environment` only has a `const` getter for its scene graph

I believe that I could get away with any of the following:
1. Merging my URDFs using xacro
2. Create a `scene_graph`, parse the various URDFs into it, and finally use `Tesseract::init`
3. By hand traverse each loaded model and add their Link / Joints to the scene

1. and 2. are suboptimal because they don't allow for online modification of the environment but 3. means writing quite a lot of code...

Ideally I would like to reach something akin to `Environment::addRobot(parseURDF(urdf))` that would allow me to add "robots" (they are static for now) to the environment.

Levi-Armstrong, 2019-10-10:
Boost graph supports the notion of subgraphs. I had planned to implement this to support your use case but have not gotten around to it. For now it would not be to much work to add a method to the environment as you suggested to that takes a SceneGraph as shown below.

```
  virtual bool addSceneGraph(const tesseract_scene_graph::SceneGraph& scene_graph);

  virtual bool addSceneGraph(const tesseract_scene_graph::SceneGraph& scene_graph, tesseract_scene_graph::Joint joint);
```

Levi-Armstrong, 2019-10-11:
Link to the boost subgraph https://www.boost.org/doc/libs/1_55_0/libs/graph/doc/subgraph.html

johnwason, 2019-11-14:
We have been able to represent multiple robots by having multiple "groups" in the SRDF file. Right now it isn't clear how you specify to concurrently plan multiple robots since the planning requests all have a "std::string manipulator" parameter. Is there a way to send a trajopt request to concurrently plan for multiple robots?

Levi-Armstrong, 2019-11-14:
Currently you create a single joint group that includes all of the joints you would like to plan for.

johnwason, 2019-12-10:
@Levi-Armstrong I tried planning with trajopt using multiple robots, but it wasn't smart enough to plan for when the arms needed to avoid each other while passing through the same space.. I will probably need to use the OMPL/TrajOpt hybrid, but it doens't have multi-robot support.

Also, the SRDFModel class is immutable, and doesn't work with SWIG because of nested classes.

Levi-Armstrong, 2019-12-29:
> I tried planning with trajopt using multiple robots, but it wasn't smart enough to plan for when the arms needed to avoid each other while passing through the same space.

Did you have continuous collision checking enabled? I would expect it not to work if using discrete collision checking.

> I will probably need to use the OMPL/TrajOpt hybrid, but it doens't have multi-robot support.

I should; what issue did you have? Multiple robot should be defined as a single joint kinematics group.

@johnwason  Do you have a test case I

PR #219 (2020-02-12): Allow to merge a SceneGraph into another one
This is a proposal to fix #142 

Basically introduce a new API to add a SceneGraph to another one, possibly with a prefix. This allows to `parseURDF` once, and add the same robot to the same scene multiple times with different names. Unfortunately, I needed to create prefixed copy operators for links and joints. As long as `name_` is a `const` member, I couldn't figure out a very good way to write them any other way.

The copying itself is done as breadth-first search over all links / joints, copying it into a prefixed version and adding to current  [... cut]
closed 2020-02-12 by Levi-Armstrong