tesseract-robotics/tesseract #143: descartes_tesseract package not found

johnwason, 2019-10-09:
The descartes_tesseract package was removed from trajopt_ros but is not in the current master branch of tesseract as expected.

haudren, 2019-10-09:
I just worked around that issue with the attached patch:

https://gist.github.com/haudren/a00d7a36481ed1f90b25240301a57b98

Basically deleted any mention of `descartes_tesseract`, and copied the necessary code into the tests. I have no idea what's the plan for `descartes_tesseract`, so this probably shouldn't make it into a PR, but allows to build successfully.

mpowelson, 2019-10-09:
I realize this is also not ideal, but you can also just checkout descartes_light before it was removed (49fdf0d8fd075954d79e9315a0796546a0d3b135). I think that PR got merged before the corresponding Tesseract one was ready. I believe [THIS](https://github.com/Levi-Armstrong/tesseract-1/tree/feature/ImproveDescartesIntegration) is the WIP Tesseract branch, but I don't know the status of it.

haudren, 2019-10-10:
Thanks for updating this issue... Unfortunately, this PR seems to be blocked by the addition of new features, is there any way to merge the minimal subset of changes needed to fix the build?

Levi-Armstrong, 2019-10-11:
This PR #141 now addresses this issue.

Levi-Armstrong, 2019-10-14:
PR #141 has been merged

closed 2019-10-14 by Levi-Armstrong