tesseract-robotics/tesseract #144: Switch to GTest imported targets?

haudren, 2019-10-09:
Currently, the code uses `GTEST_BOTH_LIBRARIES`. From a cursory look, I believe we should instead switch to using `GTest::GTest` and `GTest::GTestMain`, because `GTest` has a transient dependency on `Thread`

https://cmake.org/cmake/help/v3.15/module/FindGTest.html

I also noticed that the test code in some other repos (`opw_kinematics` at least) seems to contain typos, where the imported target is called `gtest::GTest`, which triggers the inclusion of the external project instead of using the system version.

Levi-Armstrong, 2019-10-10:
> Currently, the code uses GTEST_BOTH_LIBRARIES. From a cursory look, I believe we should instead switch to using GTest::GTest and GTest::GTestMain, because GTest has a transient dependency on Thread

I agree.

> I also noticed that the test code in some other repos (opw_kinematics and descartes_ikfast at least) seems to contain typos, where the imported target is called gtest::GTest, which triggers the inclusion of the external project instead of using the system version.

I will look into those repositories.

Levi-Armstrong, 2019-10-18:
This has been addressed in PR #149 and PR's have been opened against opw_kinematics and descartes_light.

closed 2019-10-18 by Levi-Armstrong