tesseract-robotics/tesseract #146: Windows build

johnwason, 2019-10-10:
I have patched tesseract and its dependencies to build and work with Windows. I am not sure if everything is working correctly because the current master branches are failing unit tests on Linux. The patches can be found here:

https://github.com/johnwason/trajopt_ros/tree/windows

https://github.com/johnwason/tesseract/tree/windows

https://github.com/johnwason/descartes_light/tree/windows

https://github.com/johnwason/opw_kinematics/tree/windows

The following modifications were required:

* Windows requires DLL export tags in the header files and does not like to use STL with shared libraries. For these reasons, static libraries should be used on Windows.
* M_PI is not defined by MSVC. Use EIGEN_PI instead.
* The default Gurobi solver doesn't work on Windows. qpOASIS seems to work, but OSQP is giving a "P is not upper-triangular" error.
* Use STL or Boost instead of system functions to allow cross-platform builds
* For some reason jsoncpp is not in the ROS choco repository for Windows. Using c:\opt\vcpkg to install seems to work.
* Patch https://github.com/ros/catkin/pull/1034 to catkin is required

I will leave it to the main developers to determine the best way to merge in these changes.

Levi-Armstrong, 2019-10-14:
I am going to start with opw_kinematics since it has only system dependencies. Do you know how to setup CI for windows?

Levi-Armstrong, 2019-10-14:
@gavanderhoorn I did a brief look at industrial_ci and it was not clear if ROS2/Windows is currently supported and I could not find any references in the documentation. It looks like the ROS2 support is not in the default branch only the master branch. Have you setup up a repository for ROS2/Windows using industrial_ci that I may use as a reference?

gavanderhoorn, 2019-10-14:
re: windows: no. But that would depend on whether Travis has Windows support.

re: `master` branch: the current default branch is `legacy`. ROS2 is supported in `master`. The reason for this is that up-till recently, no `.travis.yml` configs cloned a specific branch. If @ipa-mdl just changed the default branch (the ROS2 support comes with a significant rewrite of `industrial_ci`) there was the potential to break people's configurations.

Not nice, so `legacy` branch was created and made the default. After a certain amount of time, `master` will become the default branch and everyone's configs will automatically follow (ROS 1 is also supported in `master`).

As to using it to setup a Windows CI setup: that would be something @ipa-mdl should be able to help with. But I doubt it's currently possible, seeing as `industrial_ci` is `bash`-based.

johnwason, 2019-10-14:
Travis CI does support windows. It uses the Git Bash shell (the one packaged with Git) instead of the Windows command prompt. Installing ROS involves using `choco` instead of `apt-get`, but otherwise is pretty similar. It does however take a really long time to get everything installed.

Levi-Armstrong, 2019-10-14:
@johnwason Do you have an example repository I can use as reference to test on opw_kinematics?

johnwason, 2019-10-14:
I do not have an example of using ROS on Travis. I will fill in an example on opw_kinematics that you can use as reference.

Levi-Armstrong, 2019-10-14:
@gavanderhoorn Thank you, I will wait for @ipa-mdl to respond before attempting with industrial_ci.

johnwason, 2019-10-14:
Currently catkin on Windows doesn't generate bash setup scripts. This should be relatively easy to fix so I am going to look into that patch first. This will solve a few other issues I have had using the cmd prompt on Windows.

johnwason, 2019-10-14:
I was wrong, adding Git Bash support to Windows is going to be a lot of work. We will need to have a batch file called from travis.yml using cmd to run the ROS commands. I will leave an issue on catkin that this should be fixed.

johnwason, 2019-10-14:
See https://github.com/ros/catkin/issues/1035 . I don't know enough about the guts of Catkin to make this modification.

gavanderhoorn, 2019-10-14:
@johnwason: trying to use a merged install space makes sense to me.

Does that improve the situation?

johnwason, 2019-10-14:
No, that really doesn't help when I am modifying multiple packages concurrently. All that is required to support Git Bash is to generate the appropriate environment scripts. Everything else should work as it is...

gavanderhoorn, 2019-10-14:
I'm not sure I understand why that "really doesn't help"? There's also support for `--merge-devel`.

IIRC, that should result in a single entry in the `ROS_PACKAGE_PATH` instead of `N` entries for `N` packages.

johnwason, 2019-10-14:
Ok, I'll see if merging devel helps. Merging install would not change the env.bat during catkin_make_isolated

gavanderhoorn, 2019-10-14:
For an `install` space it does result in a single entry in the `ROS_PACKAGE_PATH`:

```
ROS_PACKAGE_PATH=/catkin_ws/install/share:/opt/ros/melodic/share
```

For a `devel` space I'm not sure.

johnwason, 2019-10-14:
When I run `catkin_make_isolated --merge --use-ninja` I get this error:

    '*' is not recognized as an internal or external command,
    operable program or batch file.
    <== Failed to process package 'intera_common':
    Command '['C:\\rosws\\devel_isolated\\env.bat', 'cmake', 'C:\\rosws\\src\\intera_common\\intera_common', '-DCATKIN_DEVEL_PREFIX=C:\\rosws\\devel_isolated', '-DCMAKE_INSTALL_PREFIX=C:\\rosws\\install_isolated', '-DCMAKE_BUILD_TYPE=RelWithDebInfo', '-G', 'Ninja']' returned non-zero exit status 1

The command `catkin_make_isolated --use-ninja` works until I run out of line length.

gavanderhoorn, 2019-10-14:
Why do you use `catkin_make_isolated`? Does `catkin_tools` not work under Windows?

johnwason, 2019-10-14:
`catkin_tools` does not appear to be available in the Windows package manager. `colcon` is in the package manager but I haven't used it before.

gavanderhoorn, 2019-10-14:
Might be worth a try. No support for a `devel` space though.

[migration docs](https://colcon.readthedocs.io/en/released/#migration-docs). [Building ROS1 packages](https://colcon.readthedocs.io/en/released/user/quick-start.html#build-ros-1-packages).

At least Colcon was created with Windows as one of the target platforms.

Not saying that changing `catkin` is not an option, but having tried at least these things could help convince the `catkin` maintainers that this is actually needed. Would probably also help the `ms-iot` team understand what is going on.

johnwason, 2019-10-15:
After a **lot** of trial and error I have a working sample for opw_kinematics.  This does not include the Linux build (a build matrix will be required). The solution required using two batch scripts that could be moved into a more centralized repository.

https://github.com/johnwason/opw_kinematics/tree/windows-travisci

https://travis-ci.org/johnwason/opw_kinematics/builds/598276554

Be sure to clone the windows-travisci branch

(The log messages are random, these changes should be squash merged)

Levi-Armstrong, 2019-10-16:
@johnwason Thank you I will see if I can take this and create a build matrix that handles linux and windows.

Levi-Armstrong, 2019-10-18:
@johnwason I had one question. Could the contents in the .bat file go directly in the .travis.yml file?

johnwason, 2019-10-18:
Travis CI uses Git Bash for its terminal environment on Windows. Unfortunately the vcvars files for the Visual Studio compiler env variables and the ROS env variables scripts are batch files, which are not compatible with Git Bash. Issue https://github.com/ros/catkin/issues/1035 suggested generating env scripts that would be compatible with Git Bash.

In order to call batch files, the `cmd` command needs to be used to active a standard windows command prompt and call a batch file. Any environmental variable changes are discarded after returning from `cmd` to the Git Bash. There may be a way to have these variables persist but I haven't found it yet.

To embed the batch files in the travis.yml file, the best bet is probably to just echo to a file:

    echo -e "batch_first_line\nbatch_second_line" > some_file.bat

(Using `-e` will expand \n to newlines.)

johnwason, 2019-10-18:
I have tesseract running on Windows standalone without ROS. The required DLLs are included in the repo so there shouldn't be anything else required to run beyond the MSVC redist packages.

https://github.com/johnwason/tesseract_robotraconteur_builds/tree/master/win64

Levi-Armstrong, 2020-12-11:
It appears to building on windows now. Reopen if needed.

closed 2020-12-11 by Levi-Armstrong