tesseract-robotics/tesseract #148: Return file contents instead of filename for tesseract_scene_graph::ResourceLocatorFn

johnwason, 2019-10-14:
Currently tesseract_scene_graph::ResourceLocatorFn returns the filename to a resource instead of the resource contents. This is inflexible since the resource may be stored in a zip file or retrieved from a URL. The locator function should return the file contents as a byte array. Assimp can load in the raw bytes using ReadFileFromMemory so there shouldn't be a compatibility issue.

Levi-Armstrong, 2019-10-14:
@johnwason I don't have an issue with change. @mpowelson, @schornakj  Do you see any issues with this change?

johnwason, 2019-10-14:
Can we use std::vector<uint8_t> as the return type?

mpowelson, 2019-10-15:
It would seem to me that it might be a bit of a misnomer if we changed it to return the data instead of the location. Also, currently you could use it to get the filepath of a directory to write data to in place of something like rospack. I imagine we are using it that way in a few unit tests.

Can you overload a std::function definition? Then we could have one that returns a byte array and one that returns a string. Or we could make the things using the ResourceLocatorFn be smarter and know how to parse URLs and extract zip files. Perhaps some convenient utilities are in order.

johnwason, 2019-10-15:
I plan to use a custom resource loader, so having that logic within the tesseract functions would not be great. Could be define a ResourceLoader class with "IsFile", "GetFilePath", "GetResourceData" functions?

Levi-Armstrong, 2019-10-15:
I took a look at ros resource retriever and this is what they do to address this issue. It looks like they use `std::vector<uint8_t>` when initially retrieving the data then they write it `boost::shared_array<uint8_t> data`. I assume they use a shared array to prevent from coping data. Should we return a shared pointer?

https://github.com/ros/resource_retriever/blob/kinetic-devel/include/resource_retriever/retriever.h
https://github.com/ros/resource_retriever/blob/kinetic-devel/src/retriever.cpp

johnwason, 2019-10-15:
`boost::shared_array<uint8_t>` does not contain length information. It would have to be `std::shared_ptr<std::vector<uint8_t> >` or similar.

Levi-Armstrong, 2019-10-15:
> I plan to use a custom resource loader, so having that logic within the tesseract functions would not be great. Could be define a ResourceLoader class with "IsFile", "GetFilePath", "GetResourceData" functions?

This change will work.

johnwason, 2019-10-15:
I will implement this option and issue a pull request.

johnwason, 2019-10-15:
For the custom resource loader, I am going to use the RR resource management system I am developing. This resource system will use the "buckets" concept like AWS S3 or Google Cloud Storage. Each bucket will be identified by a name/and or a UUID. These buckets can either be a directory a zip/tar file, or a remote resource service. The service definition can be found here:

https://github.com/johnwason/robotraconteur_standard_robdef/blob/master/group1/com.robotraconteur.resource.robdef

A "bucket" in this case is a generalization of a ROS package.

I need to flesh out the details of how all this will work. My plan is to redirect "package://" requests to buckets of the same name when loading URDF/SDF files.

mpowelson, 2020-03-18:
It looks like this was resolved? Should we close this issue?

Levi-Armstrong, 2020-12-11:
addressed in #172

closed 2020-12-11 by Levi-Armstrong