tesseract-robotics/tesseract #168: Clean up contact monitor

mpowelson, 2019-11-01:
The [contact monitor](https://github.com/ros-industrial-consortium/tesseract/blob/master/tesseract_ros/tesseract_monitoring/src/contact_monitor.cpp) could use a refactor. See some comments on #162.

1) It should be refactored into a class so that we can have multiple contact monitors at the same time. This should let us get rid of most if not all of the global variables there.
2) We should work to separate some of the processes more completely. For example, I would think contactTest should go in a different thread from visualization. Then that could be running for example on the GPU. Along those lines, computeCollisionReportThread currently blocks callbackJointState. I would think the joint state callback should be really quick and then when the collision checking needs them it would very quickly lock, copy the latest, and unlock the joint states.

DavidMerzJr, 2020-03-02:
@mpowelson the first point was addressed in #234.  I did not try to address the second point.

mpowelson, 2020-03-18:
Closing this as it has been moved to the tesseract_ros2 repo.

closed 2020-03-18 by mpowelson