tesseract-robotics/tesseract #174: Refactor Planner Configuration and Hybrid Planners

marip8, 2019-11-07:
The concept of the hybrid planner is to create a trajectory with a "global" planner (OMPL, STOMP, Descartes) and give that trajectory as a seed to the TrajOpt "local" planner. It seems unnecessary to create hybrid planners for each type of global planner. The hybrid planner should really be templated based on the type of global planner you want to use:

```c++
template<typename T>
class HybridPlanner : public MotionPlanner
{
public:
...
  bool configure()
  {
    // How should this method work to configure both planners generically?
  }

  bool solve(...) override
  {
    ...
    // Call global planner
    global_planner_.solve(global_planner_result);

    // Seed the local planner with the global planner result
    ...

    // Call local planner
    trajopt_planner_.solve(final_result);
    ...
  }
...
private:
  T global_planner_;
  TrajOptMotionPlanner trajopt_planner_;
}
```

This is a nice idea, but the current configuration of the planners does not allow this for a few reasons:
  - There is not a generic `configure` method in the planner base class
      - Even if there were, there is not a single set of parameters that could configure all of the different types of planners (unless we used a JSON object, XMLRPC value, or `boost::any`)
  - There is not generic planner configuration base class with which to configure the planners.
      - Even if there were, we would have to cast it to the correct derived type which does not seem like an ideal practice

We should revisit the concept of planner configuration and come up with a better solution that simplifies the interface for configuring the various types of planners in a generic way

gavanderhoorn, 2019-11-08:
Instead of using composition here (which inherently requires recompilation, or at least the way in which it is used here), it may make sense to look at implementing a pipeline architecture.

If "all" planners implement a similar enough interface, they could be wrapped in a *stage* interface, which then could be chained and called one after the other, feeding output of stage N as input to N+1.

That would make it possible to construct arbitrary pipelines.

This is exactly what MoveIt does, and also what the MTC does (and in fact, many other frameworks do).

Levi-Armstrong, 2020-09-04:
This is addressed in feature/CommandLanguage branch leveraging taskflow to build motion planning graphs

closed 2020-09-04 by Levi-Armstrong