tesseract-robotics/tesseract #176: Restructure Process Segment

Levi-Armstrong, 2019-11-10:
Current the process segment class has a std::vector for the approach, process and departure.
Usually the there are duplicate waypoints because the end of the approach waypoint is also the start waypoint of the process. This requires a lot book keeping when modifying the segment for example removing failed vertices from descartes or adding new vertices at the end of the approach. 

I propose changing the class to maintain a single vector for all waypoints and then provide functionality for inserting, removing, approach iterator, process iterator, departure iterator etc for each.
```
segment = {1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15}

approach = {segment.begin(),  segment.begin() + 5}
process = {segment.begin() + 5, segment.begin() + 10}
departure = {segment.begin() + 10, segment.end()}
```

mpowelson, 2019-11-11:
I feel like there may be another way to skin this proverbial cat. The current structure seems pretty nice to me. For one, it allows the use of your generators. If we went to a flat list then if you had a massive process, you'd have to copy all of the data to change the approach, but then again maybe that's not so bad. I suppose a flat list would make reversing the strips marginally easier.

The same effect to what you are proposing would be to add some sort of identifier or message to the waypoint where someone could give them whatever categories they wanted. Ours would just happen to be approach, process, and departure. 

Perhaps the easier thing to do would be to add  a removeProcessPoint function or something that then handles the removal of the duplicate points as well. However, I may have to think about it some, but if the endpoints are duplicated, wouldn't they both fail descartes? Then when you removed all failed won't they both get removed anyway?

Levi-Armstrong, 2020-09-04:
replaced by command language

closed 2020-09-04 by Levi-Armstrong