tesseract-robotics/tesseract #181: Don't use tesseract_scene_graph::Link and tesseract_scene_graph::Joint for parameters in environment.h

johnwason, 2019-11-20:
Currently `tesseract_scene_graph::Link` and `tesseract_scene_graph::Joint` are used to pass parameters to functions in environment.h. This will cause an unnecessary copying. Instead pass by reference or use shared pointer types.

Example:

https://github.com/ros-industrial-consortium/tesseract/blob/60e4cedc307dcfa0d514a835aaf6c7d4b004ef60/tesseract/tesseract_environment/include/tesseract_environment/core/environment.h#L147

Levi-Armstrong, 2019-11-30:
@johnwason Thank you. I am currently going through and adding Clang-Tidy to both tesseract and trajopt and I have PR #183 which addresses this along with several others like it.

Levi-Armstrong, 2019-12-26:
Addressed by PR #183

closed 2019-12-26 by Levi-Armstrong