tesseract-robotics/tesseract #184: Support glTF meshes

johnwason, 2019-12-03:
A new mesh format has been released and is rapidly gaining traction:

https://en.wikipedia.org/wiki/GlTF

https://blog.ybalrid.info/2018/07/why-gltf-2-0-is-awesome/

It appears to be supported by assimp. This new format is intended to fix the problems with collada, stl, etc.

gavanderhoorn, 2019-12-04:
One more standard to rule them all :)

When I looked at it some years ago it was almost as if someone had dusted off VRML and gave it a new name.

johnwason, 2019-12-04:
I guess the file types will continue to proliferate until one isn't terrible...

This one apparently matches the format used by WebGL, with the goal of minimizing the processing required to load into the GPU.

Levi-Armstrong, 2019-12-05:
If it is supported by assimp then the file should get processed if added to the urdf.

johnwason, 2020-04-21:
An interesting article discussing glTF from the Godot team:

https://godotengine.org/article/we-should-all-use-gltf-20-export-3d-assets-game-engines

glTF 2.0 supports vendor extensions:

https://github.com/KhronosGroup/glTF/tree/master/extensions

Would it make sense to add robot specific extensions to glTF as a replacement for SDF and URDF?

gavanderhoorn, 2020-04-22:
It would be interesting, but this starts to sound rather like Collada. That also supports extension points.

A good example of a tool which uses extensions to Collada heavily for robotics purposes is OpenRAVE.

Problem with "vendor extensions" is that most of the time, vendors don't support extensions of other vendors (for whichever reasons). You can then get into the situation where loading a mesh with extension `vendor_A` and serialising it again with a tool from `vendor_B` would result in loss of the elements in the mesh which use/depend on the extension. As `vendor_B` has decided not to add support for that extension.

I've not checked whether Khronos has made provisions for this in the standard (ie: require `in==out` for unsupported extensions), so it could be they've plugged this hole, but at least for Collada, what I describe above happened.

---

And it was exactly the extension system which was one of the biggest issues with Collada when it was discussed as a potential successor to urdf.

gavanderhoorn, 2020-04-22:
We could of course decide we care about none of that and still use glTF. I did not want to imply that should not be done.

But if the goal is to find a format which can be used both as urdf / sdf and what have we can (ie: by humans, in OSS projects, etc) *and* can be reliably imported and exported from commercial tools, we have to be careful.

johnwason, 2020-04-22:
@gavanderhoorn Khronos is collecting extensions, reviewing, and storing them in the main repo:

https://github.com/KhronosGroup/glTF/tree/master/extensions

There is a dependencies statement in the file to indicate which extensions are required.

It also appears Solidworks has an exporter for glTF now: https://blogs.solidworks.com/solidworksblog/2019/01/introducing-solidworks-extended-reality.html

Blender and Autodesk also support it.

I am not sure about how human readable the file will be. This is something that will need to be investigated before using it.

gavanderhoorn, 2020-06-06:
Seems this discussion is something which could be interesting in the context of https://github.com/ros/urdf/issues/34 as well.

gavanderhoorn, 2020-06-06:
And I just noticed Blender 2.8 and newer support gTLF.

johnwason, 2020-06-06:
There are a few existing proposals to create a glTF based replacement for URDF and SDF:

* https://ieeexplore.ieee.org/abstract/document/8971739
* https://gist.github.com/mikedh/ce09d1bf83767fefded25be519a61564

I am probably going to try to put together a working group to see what interest there is. I don't think either URDF or SDF are going to work for my current projects.

gavanderhoorn, 2020-06-06:
> I don't think either URDF or SDF are going to work for my current projects.

I'd suggest to try and reuse something existing if at all possible, for obvious reasons.

johnwason, 2020-06-06:
We are planning on using glTF with extensions

gavanderhoorn, 2020-06-06:
> There are a few existing proposals to create a glTF based replacement for URDF and SDF:
> 
>  * https://ieeexplore.ieee.org/abstract/document/8971739
>  * https://gist.github.com/mikedh/ce09d1bf83767fefded25be519a61564

If we're honest, I'd hardly call these proposals. The first is a paper by two researchers and the second is a gist with some good ideas, but there doesn't appear to be any implementation available (and the gist is already almost 1.5 years old).

You seem to have started KhronosGroup/glTF#1816. Is that the "glTF with extensions" you mention?

johnwason, 2020-06-06:
Yeah, I asked @wlawler45 to take a look. The amount required to add robot information to glTF is fairly minimal. We are going to come up with a formal proposal over the next few weeks/months. The infrastructure to support glTF is already widely available (far more than even COLLADA), so it should not be all that difficult.

gavanderhoorn, 2020-06-07:
There are now a few different discussions going on in parallel.

I posted a link to this issue here on ROS Discourse: [Web-based URDF visualization tool and library opensourced from NASA JPL](https://discourse.ros.org/t/web-based-urdf-visualization-tool-and-library-opensourced-from-nasa-jpl/14404/5).

@gkjohnson responded there. Perhaps he'd like to post a few comments here as well.

gkjohnson, 2020-06-08:
Thanks @gavanderhoorn! I'm happy to share a few thoughts.

For some more context I work at JPL as a software engineer specializing in robotics visualization tooling but these are of course just my personal opinions. While we do quite a bit of ROS related work we primarily launch non-ROS-based platforms so my use cases may reflect a different point of view but I'm a big advocate for using standards for our projects. About 6 or 7 years ago I started pushing for using URDF within our work so we could take advantage of a standard format pioneered by the robotics field and let us easily swap between robots in our non-ROS tools. I've since [open sourced some work](https://github.com/gkjohnson/urdf-loaders/) for loading, rendering, and manipulating URDF files independent of the ROS ecosystem.

I'm sure there's a lot of background here I'm not privy too so please do fill me in but I'm a bit disappointed to hear there's work on _new_ formats for addressing this issue rather than fixing the problems associated with an existing format. I think fundamentally I don't understand the desire to put every piece of information into a single file. Most of downsides in the gist proposal I can see seem to revolve around too many separate files but in my opinion the inability to do this is a shortcoming of GLTF. Why can't I reference a reusable sensor model as a different file? Having URDF point to external geometry formats is useful! As it is URDF can evolve and reference any mesh format without having to change as the graphics world advances. By buying entirely into a specific geometry format you're taking away the ability for people to choose what best suits them. Since GLTFs development Cesium (one of the groups that helped develop GLTF) has created _4 new formats_ while making 3d tiles to address problems that GLTF doesn't solve: B3DM, I3DM, PNTS, and CMPT. And two of those wrap GLTF. My point is chasing a specific geometry format is not a winnable game. COLLADA was not that last graphics format and GLTF won't be either.

But don't get me wrong! I like GLTF quite a bit and it solves a lot of problems -- I'm using it in quite a few projects including for our URDFs but I want the ability to choose my geometry format based on my needs. I think GTLF and mesh formats should be left to do what they're intended to be good at -- storing geometry data -- and the robotics community should own and control a format that addresses robotics problems.

Regarding URDF there are issues with it but nothing that I don't think it can't evolve in to. It's a well-scoped, easy to understand format. I just want a consistent way to store the kinematic structure of my robot. I've been looking for movement on URDF2 for awhile now to address one my biggest gripes: lack of support for parallel kinematics, but it doesn't seem to have gone anywhere. To that end the tree-structure hierarchy baked into GLTF does not intuitively lend itself to supporting this type of structure. If URDF isn't desireable for some reason I'd hope that whatever new format is developed can retain its positive attributes.

I'm curious as to what the motivation is for a new format. I know my use cases may be considered a bit fringe but I think if any community is in a position to push a good, universal standard for solving this sort of thing it's this one. I fear if there isn't some sort of consistency in this direction we'll have to continue developing some custom formats and variants which I'd really not like to do. I'd definitely be willing to be convinced that GLTF integration will solve more problems than it creates, though. I'm just not seeing it yet.

I'm sure this is a much larger discussion than just this thread if there's interest I'd be happy to talk about it further if there's another venue. Sorry if it's a bit of a rant but thank's for hearing me out! And thank you guys for putting time into the community to enable so much of the work that's come out of it.

gavanderhoorn, 2020-06-08:
@gkjohnson: thanks for the comments here.

It seems some of the glTF developers/community members share your views: https://github.com/KhronosGroup/glTF/issues/1135 seems to go in a similar direction (https://github.com/KhronosGroup/glTF/issues/1135#issuecomment-640714609).

johnwason, 2020-06-08:
@gkjohnson I interned at the DARTS lab at JPL back in 2004-2005, not sure if it is the same group

The primary motivation for the RPI and my group to develop something different is for the new Robot Raconteur ecosystem that is slightly different than ROS, in that the different pieces of information will be distributed among the components in a more ad-hoc fashion rather than being on the same file space like in ROS. A custom glTF extension just for RR and/or ROS will be added that handles collecting the pieces and assembling the full glTF scene based on RR and/or ROS identifiers. The glTF pieces can then be assembled and loaded into various engines like Babylon.js or Godot without needing custom plugins, it would just be a matter of stitching together the glTF file pieces that come from other locations. Loading physics will require defining standard rigid body physics, but that seems to be underway. The rest of the world is moving toward glTF rapidly, so I would rather use extensions to the globally used format rather than something niche for the small open source robotics community.

gavanderhoorn, 2020-06-09:
> The rest of the world is moving toward glTF rapidly, 

As a mesh format, yes.

> the different pieces of information will be distributed among the components in a more ad-hoc fashion rather than being on the same file space like in ROS

So is the "resources have to be local" challenge the real motivation for looking at glTF?

Because that sounds like something which could (and perhaps *should*) be solved differently.

> The glTF pieces can then be assembled and loaded into various engines like Babylon.js or Godot without needing custom plugins, it would just be a matter of stitching together the glTF file pieces that come from other locations.

If the current planned approach would be to do this with an extension to glTF, how would that be different from 'custom plugins'? As far as I can tell, there is no requirement for glTF consumers to support specific extensions, is there? Not as long as an extension is made part of the spec at least.

> Loading physics will require defining standard rigid body physics, but that seems to be underway. 

From the comments on KhronosGroup/glTF#1135 it would seem using glTF as a storage format for arbitrary data is not something which is necessarily considered best-practice.

gkjohnson, 2020-06-10:
> @gkjohnson I interned at the DARTS lab at JPL back in 2004-2005, not sure if it is the same group

Hey hey! I suppose I shouldn't be surprised to run into someone who used to work at JPL on here. The group I work in focuses on tools specifically for commanding and operations rather than simulation -- I'm familiar with the DARTS group, though.

I agree with @gavanderhoorn. I feel the need for an application to dynamically derive the robot model (which sounds awesome) is separate from the need to create a new format and is something that can be solved differently. A new format being popular is not a reason on it's own to use it. It's still not clear to me why something like URDF can't be generated and used to describe the system for this use case.

I'm also not sure if I'm so confident I see a future where everything just works because all the information has been stuffed into GLTF. Someone will have to implement and maintain the extensions in those platforms and it will probably have to be those in the robotics community if they're the ones interested in using it. That aside pushing this into GLTF as an extension means involving an entirely new organization (the GTLF standards folks) which I can only see bringing more bureaucratic complexity into updating and evolving the standard when considering the needs of the robotics community. GLTF should be responsible for graphics and the robotics community needs to be responsible for the formats and data needed for our work. I don't see pushing standards on to GLTF taking responsibility off of the robotics community -- it possibly just adds more.

I'm still not really seeing positives here and if there are any they seem far outweighed by the negatives and loss of features I've mentioned previously. GLTF might be popular right now but it should be considered in the context of the pros and cons that come with the decision and I don't see the cons aren't being addressed at all.

I think I'd rather understand the apparent failings of the current system so a solution can be discussed that doesn't come at the cost of all the other good qualities of the current state of the art robotics formats that are being used. It's possible I don't understand that problem you are trying to solve yet.

gavanderhoorn, 2020-06-10:
@johnwason: it may appear as such, but I'm not necessarily against using glTF here (and it wouldn't matter if I was, I'm not involved in your project).

My motivation for posting my comments is to better understand your use-case, and to see whether there are ways to achieve what you're after without going for a different graphics format immediately.

As an example: to deal with the "mesh resources have to be local" aspect, Siemens created a `file_server` node ([here](https://github.com/siemens/ros-sharp/tree/d3f6497a69ac114db87fce1aea976b31122f7acf/ROS)) as part of ROS Sharp which provides access to `package://` URIs over a websocket. This now allows them to load `.urdf`s with `package://` URIs in them on Windows, with all the resources still hosted elsewhere.

They also had one which served the same resources over plain HTTP -- similar to how RobotWebTools does this.

Running such a server as Siemens created would be minimum overhead while almost transparently solving the remove assets problem you describe.

It doesn't address (recursively) composing scenes out of distributed descriptions of course, but that would be a different problem.

gavanderhoorn, 2020-06-11:
> There are a few capabilities need to have asap (my current project ends in a few months):
> 
> 1\. Model hierarchy and namespacing / scene definitions

This is indeed not something URDF supports right now. SDF does, but you've ruled that out as too complex.

> 2\. Including other files from network sources to build scene

This would not be something you solve by using a different format imho. See the earlier comments in this thread.

Another way of saying this: there are perhaps alternative ways to solve this.

(and off-topic, but Collada actually support(s)(ed) composing models out of other models and supports `http` and other URI schemes ;) )

> 3\. Quaternion orientations and poses

This is actually something which can be supported now that there is a URDF version attribute: https://github.com/ros/urdfdom/pull/123.

Perhaps @clalancette could add something here.

> 4\. Use glTF for meshes instead of STL or COLLADA (both outdated, COLLADA is not used correctly in URDF causing incompatibilities)

If Assimp supports it, you can use it in URDF.

> 5\. Simplicity to parse for web based systems (XML requires a lot more code to parse than glTF)

There are various projects which load URDF in javascript libraries (such as [ros3djs](http://wiki.ros.org/ros3djs/Tutorials/VisualizingAURDF) and [roslibjs](http://wiki.ros.org/roslibjs/Tutorials/UrdfParsing)). You cannot use those?

@gkjohnson seems to also be able to load them.

> 6\. Maximum compatibility with other software

This is rather vague, as this is not a metric which is fixed in time.

VRML was supposed to be a format which achieved this. Then we got Collada to do that. Then we got x3d and now glTF. But pragmatically: glTF certainly has more support than URDF.

But again: URDF is not a mesh storage format. glTF is. It's like wanting to use URDF with 3D printing tools: it just doesn't store the same information.

> glTF does everything on this list except for including other files, which can be accomplished with a small extension and preprocessor. 

It's obviously completely up to you to do whatever you want/feel necessary. Including arbitrary content in glTF seems supported, but from reading the issues on the Khronos tracker it seems it's not something the devs there would seem like the right thing to do.

> I am open to other options, but none seem to be appearing. I have been asking about URDF2 for several years without any movement in that direction, 

It's going to sound pedantic, I realise, but "asking" is not the right way to go about this in an OSS community. If the currently available solution works for the use-cases of the majority of the users, people who want something else will have to work for and on it. Otherwise it's not going to happen.

The situation with URDF has been awkward, so even with putting in the work things didn't change unfortunately.

We can only hope that improves now, but that's a bit weak.

---

Thanks for writing down your motivation and requirements. I've commented on some of them, but I realise you will have additional constraints and requirements which may make glTF the perfect format. That's obviously OK, as I wrote earlier: I'm not involved in your project.

gkjohnson, 2020-06-12:
Thanks for a more detailed list.

I am using URDF with GLTF very successfully in web tools but it sounds like you're interested in using GLTF for more than URDF is intended for -- to represent and transmit / sync state for everything in system rather than just robots.

To a few of your points:

> 3. Quaternion orientations and poses

I'm not following why you need this in the file format itself. It should be easy to convert to and from euler angles. If you're trying to keep two networked states in sync then transferring quats between clients shouldn't need to affect the format used to describe the scene or robot in the first place. I could agree it makes it slightly easier, though.

> 4. Use glTF for meshes instead of STL or COLLADA (both outdated, COLLADA is not used correctly in URDF causing incompatibilities)

Related to the [discussion that @gavanderhoorn linked](https://discourse.ros.org/t/web-based-urdf-visualization-tool-and-library-opensourced-from-nasa-jpl/14404) I've updated the spec wiki to mention that other formats are valid. There's no reason to limit yourself to pointing to STL or COLLADA files when making a URDF. RViz supports more formats and likely GLTF, as well, by using Assimp.

> 5. Simplicity to parse for web based systems (XML requires a lot more code to parse than glTF)

This I don't necessarily agree with. Browsers have built in XML parers that are fast and contain useful query helpers. I have built both a [urdf parser](https://github.com/gkjohnson/urdf-loaders/tree/master/javascript) and [xacro processor](https://github.com/gkjohnson/xacro-parser) for the web and most of the code needed for the URDF parser is what would be needed to add joint articulation to any GLTF model. GLTF is not necessarily a simple format to parse unless you're referring to the idea that someone's already done it for you.

It sounds like you're looking for a common "scene" format, which is not something that I've really seen done I don't think because the needs and capabilities of each platform differ so much and benefit from different representations. In my opinion this should be a sort of extendable "meta" file specifically so it can point to different assets in different formats and use them in the same environment together. How will you store heightmaps? Or point cloud information? Especially in proprietary formats? Will everything have to be converted to a mesh? It's useful and more traceable when data is represented in it's original format.

I definitely not my intent to derail your project especially if it has a short timeline. I'm just hoping to help move this type of format in a direction that is agnostic to platform and evolve-able independent of the graphics field. For my own work I don't think the proposed format would be use-able. If I needed a robotics scene format I'd probably look at something like SDF myself but it sounds like that's not appealing.

Almost 100% of my projects are web-based tools, now, so I've had to deal with some of this before at least. I don't know what kind of web based engine you're using or if this is helpful but I've open sourced a URDF loader for three.js [here](https://github.com/gkjohnson/urdf-loaders/tree/master/javascript) that can support loading any geometry format. If new features are needed at all I'm happy to help get them in place.

Regarding URDF2 I'm not sure why it hasn't moved, who to talk to about that, or if there's a proper venue to discuss it. Some of the features listed in the proposal wouldn't affect existing parsers so it might just be a matter of getting them into the current spec and letting it get implemented where needed.

Thanks again for taking your time to talk through this!

johnwason, 2020-06-12:
I think between @gavanderhoorn @gkjohnson @Levi-Armstrong @wlawler45 myself and maybe someone from ARM and/or NIST we probably have enough people for a working group. Shall we set one up to hammer something out? I am open to other options beyond glTF but we need something to fill this gap.

gkjohnson, 2020-06-15:
@johnwason I'm happy to take part if you want to set something up!

johnwason, 2020-06-18:
We discussed this a bit during our project meeting, and @Levi-Armstrong suggested we create a URDF 2.0 spec and begin using it with tesseract. Maybe we should call it urdf-ng (next gen) to distinguish it from the OSRF URDF. I am open to this option if that is more agreeable.

johnwason, 2020-06-18:
@gkjohnson do you think it is possible to integrate something like this editor with a URDF system? https://threejs.org/editor/

I have been planning to use Babylon.js because it has an editor, VR, and physics engine directly integrated. Is there a reason you chose three.js instead of Babylon.js?

gkjohnson, 2020-06-20:
> @gkjohnson do you think it is possible to integrate something like this editor with a URDF system? https://threejs.org/editor/

I'm less familiar with the three.js editor but looking over the code briefly it would require modifying the editor to add something like URDF into the import dropdown but that should be doable. Maybe a sidebar could be added to display joint angles.

> I have been planning to use Babylon.js because it has an editor, VR, and physics engine directly integrated. The tesseract_viewer_python in this repo uses Babylon.js. Is there a reason you chose three.js instead of Babylon.js?

I use and contribute to three.js quite a bit and have less experience with Babylon so full disclosure my answers will likely be skewed by that a bit. Realistically this can be done in any engine so it really comes down to personal preference.

For me the three.js codebase is a bit easier to read and the architecture more manageable to grok which is important for making changes I need for my work. Three.js is pretty unopinionated and focuses on providing a renderer / visualization tool around which an application can be built. Personally I've moved away from monolithic "do everything" packages because it removes a lot of control during software development and comes with a lot of unneeded features (for our case). It can be nice at first but inevitably you hit a wall in terms of what it's designed to do. I've worked with other game engines in the past and while it's quick to get started it's frustrating when you want to do something outside the scope of the package. Not that three.js doesn't have it's own frustrations I've just found them more manageable. To that end I don't really need a physics engine for my work, either. Our telemetry is derived from real robots or purpose built simulators that use the robot software in the loop. In the case that you do need a real time physics engine for your work there are a couple out there (ammos.js based on Bullet, and cannon.js) that you can integrate with three but I'm not sure how straight forward it is. Regarding VR / XR three.js keeps up to date with the spec and supports it out of the box, as well. I also find the three.js community to be pretty supportive if you run into an issue but of course I can't speak for Babylons -- I'm sure it's great, as well.

Again those are all really just personal opinions. One thing to note Babylon.js is that it does use a left handed coordinate system by default. It looks like you can change it but it's something to keep in mind if you use it (and it looks like this repo is already doing so).

If someone would like to help convert the URDFLoader I've linked to abstract the URDF loading component a bit so it can be repurposed for Babylon js, as well, I'd be happy to guide some of that in the repo.

> suggested we create a URDF 2.0 spec and begin using it with tesseract. Maybe we should call it urdf-ng (next gen) to distinguish it from the OSRF URDF. I am open to this option if that is more agreeable.

That sounds great. URDF has long needed a "version" field so that would be a fine way to differentiate to start. It sounds like you might need some more expansive changes but for my use I'm really just interested support for parallel kinematics. It would be nice to have a recommended algorithm for resolving closed loop joints for consistency, as well.

gavanderhoorn, 2020-06-20:
> for my use I'm really just interested support for parallel kinematics.

The "only" reason URDF (as in: the file/markup format) doesn't allow you to have joints with multiple parents is because the default (and essentially only) IK solver at the time it was created didn't support those (ie: KDL).

> It would be nice to have a recommended algorithm for resolving closed loop joints for consistency, as well.

Allowing multiple parents in URDF is really not that difficult, and just requires a few changes.

It's the algorithm behind it that is where the work is (finding one that is).

clalancette, 2020-06-22:
I have a few thoughts to share here.  None of this is meant to discourage the current work going on, but is more intended to give some ideas of what has been going on so we can better collaborate.

1.  We did recently introduce the "version" tag to URDF.  That was done in https://github.com/ros/urdfdom/pull/133 and https://github.com/ros/urdf_parser_py/pull/52 .  With that in place, it may be possible to now extend the "original" URDF with newer versions, without forking it completely.  That is, we could define version 1.1 of the URDF spec with new features that people wanted.
1.  If there is sufficient interest by some parties here, it may be possible to extend the maintanership of URDF beyond just Open Robotics people.  I'd have to discuss this with my co-maintainers @scpeters and @sloretz, but please let me know if there is interest in doing so.
1.  At the last ROSCon we had an unofficial Birds-of-a-Feather session where we talked about the future of URDF.  Two main things came out of that discussion: a) URDF needed a version tag to continue development, and b) that it may make more sense to switch to a different, existing format completely.  The major ones that were brought up were SDF and COLLADA.  Expanding the ROS ecosystem to accept SDF is being heavily considered and prototypeed in https://github.com/ros/urdfdom_headers/issues/59 .  So it probably makes sense for this effort to join with that one.  It would be particularly unfortunate if we ended up in the case where this thread of work ended up defining a "URDF 2.0" that duplicated a lot of the features of SDF.

Anyway, those are my thoughts.  Please let me know what you all think.

mpowelson, 2020-06-22:
> 1. We did recently introduce the "version" tag to URDF.  That was done in [ros/urdfdom#133](https://github.com/ros/urdfdom/pull/133) and [ros/urdf_parser_py#52](https://github.com/ros/urdf_parser_py/pull/52) .  With that in place, it may be possible to now extend the "original" URDF with newer versions, without forking it completely.  That is, we could define version 1.1 of the URDF spec with new features that people wanted.

From my perspective, one of the biggest problems we had when @Levi-Armstrong made the Tesseract URDF parser was that there wasn't really a defined URDF spec other than the wiki. I'd have to dig through the [pull request](https://github.com/ros-industrial-consortium/tesseract/pull/117) to remember the specific issues we had, but there were several occasions where the wiki was unclear on things. It seems to be more written towards how to write a simple URDF and not how to write a parser that handles all edge cases in a uniform way. We basically just had to reverse engineer the requirements from urdf_dom.

Adding a version to the spec is great, but in order for it to be really useful I would think that the meaning of each version of the spec has to be clearly documented somewhere such that users can say they have a version xx.yy compliant parser with some confidence and know what it means to update their parser to support the next version.

traversaro, 2020-06-22:
> Adding a version to the spec is great, but in order for it to be really useful I would think that the meaning of each version of the spec has to be clearly documented somewhere such that users can [...]

A related proposal is presented in https://github.com/ros/urdfdom_headers/issues/62 .

Levi-Armstrong, 2020-06-22:
@clalancette We are interested in collaborating on further extending URDF mostly around the specification as for the near future we will maintain our own parser. I am interested in getting some of the current features in our parser incorporated into the specification like quaternions, convexhull geometry, octomap geometry, capsule geometry, etc. I would also like to see some of the concepts from the SDF format like models incorporated into the URDF specification over using xacro.

johnwason, 2020-06-22:
Whichever file format is selected, it needs to have a strong extension system like glTF, which makes it clear which extensions are in use and whether these extensions are optional. There should probably also be an "alt" tag for if the extension isn't available. Trying to include every special use case in the URDF spec is probably not necessary, and trying to work around the lack of extensibility is leading to incompatibility.

clalancette, 2020-06-22:
> From my perspective, one of the biggest problems we had when @Levi-Armstrong made the Tesseract URDF parser was that there wasn't really a defined URDF spec other than the wiki.

To be clear, there is a spec: https://github.com/ros/urdfdom/blob/master/xsd/urdf.xsd .  I don't know whether it is 100% up-to-date, or fully supported by the parser, but it should be mostly correct.  That being said, there is of course a lot of room for improvement, to the spec itself (to add new features), to the parsers (to support existing or new features), and to documentation.

To my mind, though, the documentation aspect alone doesn't warrant a new specification.

And just to be totally clear; I don't really care which way we go here.  I'm not an expert in this area.  But I do want to make sure all of the parties that are interested in this (and there are a lot of them), are talking to each other.

clalancette, 2020-06-22:
Thinking about it further, I think the right path forward here would be to put together a working group, something like "Robot/World Model Representation".  This would be a working group similar to the Real-time Security, Tooling, or myriad other groups.  That way, it would be announced on https://discourse.ros.org, and a variety of interested parties can interact in a more real-time fashion.

What do people think of this?  Is there someone willing to start up and run a working group like this?

johnwason, 2020-06-22:
Another option is to create a subset of SDF. SDF has many features which are Gazebo specific, or not really necessary for robot description.

I suggested a small working group earlier. I am on a deadline of a few months for a different project that doesn't necessarily need full ROS compatibility, although having something universal is certainly preferred to avoid fragmentation. The general ROS community has been very slow in this area so I don't want to get bogged down. I think the best path forward is to have a small group figure out a workable solution with well defined extensibility for future or niche needs.

johnwason, 2020-06-22:
Looking through the specs of URDF and SDF, we are basically going to reinvent SDF to add the necessary functionality to URDF, with a few relatively minor features like quaternion support and extension enforcement added on.

Levi-Armstrong, 2020-06-22:
> What do people think of this? Is there someone willing to start up and run a working group like this?

I am onboard.

Levi-Armstrong, 2020-06-23:
> Looking through the specs of URDF and SDF, we are basically going to reinvent SDF to add the necessary functionality to URDF, with a few relatively minor features like quaternion support and extension enforcement added on.

I agree.

johnwason, 2020-06-23:
This is a document explaining one of the projects motivating this need (currently funded):
[Open-Source Teach Pendant Programming Environment Software Architecture.pdf](https://github.com/ros-industrial-consortium/tesseract/files/4816465/Open-Source.Teach.Pendant.Programming.Environment.Software.Architecture.pdf)

gkjohnson, 2020-06-23:
@gavanderhoorn

> IK solver at the time it was created didn't support those (ie: KDL)

You probably know more about the state of RViz / the ROS ecosystem more than I do -- has the solver been updated? Would applications be able to handle systems with parallel kinematics? Is this a matter of the docs not being updated, as well?

@mpowelson

> Adding a version to the spec is great, but in order for it to be really useful I would think that the meaning of each version of the spec has to be clearly documented somewhere such that users can say they have a version xx.yy compliant parse...

Agreed. I think documentation is a big need especially if it's expected that the format can be used more broadly.

@johnwason

> Whichever file format is selected, it needs to have a strong extension system like glTF, which makes it clear which extensions are in use and whether these extensions are optional.

Definitely +1 to a proper extension or "custom data" capability. 

> Looking through the specs of URDF and SDF, we are basically going to reinvent SDF to add the necessary functionality to URDF, with a few relatively minor features like quaternion support and extension enforcement added on.

This is a bit of what I'm afraid of. I think a distinction should be made between reusable robot model and the context / scene in which it's operating. My feeling is that URDF should represent a built robot and the properties associated with it -- sensors, joints, geometry, etc -- and should be kept separate from the environment in which said robot might be operating or simulated because it can be operated elsewhere. A scene should be defined and it should point to the assets representing the robots operating in it. To that end it feels that things like an occupancy map, point clouds, or terrain should be represented in the "scene" description rather than the robot description. Perhaps SDF is that scene format (I'm trying to find some time to learn abit more about it) but there seem to be dissenting opinions, as well. Maybe creating a more distilled or general version of SDF (or moving the existing format in that direction) would be good.

@clalancette
> To be clear, there is a spec: https://github.com/ros/urdfdom/blob/master/xsd/urdf.xsd . I don't know whether it is 100% up-to-date, or fully supported by the parser, but it should be mostly correct.

That's a schema to help validate the contents of a file but when I think of a specification I expect documentation that can help me interpret a file. I can't know how the model is supposed to be structured or read based on that xsd -- you can't implement a proper loader for the robot from that, for example. I think up to date documentation is important if we want the format to be usable at or interpret-able in custom tools.

> What do people think of this? Is there someone willing to start up and run a working group like this?

I'd be happy to join!

Levi-Armstrong, 2020-06-23:
> This is a bit of what I'm afraid of. I think a distinction should be made between reusable robot model and the context / scene in which it's operating. My feeling is that URDF should represent a [...]

I agree and what you are describing is more or less the [SDF format](http://sdformat.org/spec). Where in this case the model element in SDF is the the URDF as you described it above. Then the world and light tag is the scene representation.

johnwason, 2020-06-23:
I created a repo for a potential working group: https://github.com/wasontech/robot-description-wg

For now I am using my company's organization. This may move to RPI or ARM depending on how a formal group is organized. Please create one issue for each feature request so we can discuss them individually.

johnwason, 2020-06-24:
Please watch https://github.com/wasontech/robot-description-wg if you wish to be notified of new proposed changes.

clalancette, 2020-06-25:
> Please watch https://github.com/wasontech/robot-description-wg if you wish to be notified of new proposed changes.

I'm also going to setup an in-person call and announce a working group on discourse.  I just have been running out of time every day.  I'll post here once I've set something up.

johnwason, 2020-06-25:
@clalancette I think it may be better to create an initial proposal first with a small group before announcing to the broader community. These things have a tendency to get bogged down with too many cooks in the kitchen.

johnwason, 2020-06-25:
Is there a charter template for ROS 2 working groups?

johnwason, 2020-06-25:
Initial charter added to readme: https://github.com/wasontech/robot-description-wg/blob/master/README.md

A two step process is suggested to help deal with deadlines for our active projects.

clalancette, 2020-06-25:
> @clalancette I think it may be better to create an initial proposal first with a small group before announcing to the broader community. These things have a tendency to get bogged down with too many cooks in the kitchen.

While I somewhat agree, there is a lot of duplication going on here.  Which means there are already a lot of cooks, they are just in different kitchens and not talking to each other.  That's why I'd rather have all of the people who care about this get together at least once.  If the outcome of that is a small group that goes away and comes back with something, that's fine.

johnwason, 2020-06-29:
@clalancette that seems reasonable. If we want to do a community-wide working group, it should probably be done through ROS rather than me running it (I am a one-man shop). That unfortunately does not solve the active projects' near-term needs.

closed 2020-12-11 by Levi-Armstrong