tesseract-robotics/tesseract #186: Introducing tesseract_viewer_python

johnwason, 2019-12-09:
For the projects we (RPI and wasontech) are working on, we need a geometry and trajectory viewer that runs inside a browser. Thanks to recent advances on WebGL and WebVR/WebXR, browsers are capable of displaying 3D renderings with reasonable performance. There are several rendering/game engines designed to run in the web browser. 

`tesseract_viewer_python` is a proof-of-concept of using WebGL  Babylon.js to display geometry and trajectories from Tesseract. It works by starting a small Python web server that serves the software and trajectories to a web browser. This seems to work reasonably well. The main limitation is that the browser needs to poll to receive updates, and requires a reload to change geometry. This will be improved in the future.

Thanks to Bablyon.js's built-in support, the viewer is VR capable. Use the Firefox Reality browser to view the webpage on a headset, and click the goggles icon on the lower right. This will start the VR experience.

The source code:

https://github.com/johnwason/tesseract_viewer_python

A generated example (just click the link to view!)

https://johnwason.github.io/tesseract_viewer_python_example1/

(I am aware of ros3djs. We want to use Babylon.js)

Levi-Armstrong, 2019-12-10:
This is cool! Are you interested in integrating this into tesseract along with the python wrapper; I am interested?

johnwason, 2019-12-10:
Yeah, I can create a pull request. This is very crude though, and I am not planning on improving it beyond fixing major bugs. My plan is to create a viewer for the RR ecosystem based on Babylon.js, which will work with the RR tesseract server.

jf---, 2019-12-10:
> (I am aware of ros3djs. We want to use Babylon.js)

It would be cool if a matplotlib style architecture could be adopted, where you have multiple backends ( Qt, threejs, babylon.js ) , but a unified API

> the RR ecosystem

Sorry, I couldnt deduce from the above what you're refering to, but curious for sure ;)

johnwason, 2019-12-10:
@jf-- we are using tesseract for these projects:

http://arminstitute.org/projects/robot-raconteur-rr-an-interoperable-middleware-for-robotics/
http://arminstitute.org/projects/open-source-teach-pendant-programming-environment/

(Robot Raconteur is the flagship product for my (one man) startup)

johnwason, 2019-12-10:
It is also possible to create an application using Electron (https://github.com/electron/electron) or CEF (https://github.com/cztomczak/cefpython). I am not planning on upgrading this simple viewer to use these options, but it may be of interest in the future.

Levi-Armstrong, 2020-09-04:
This has been merged.

closed 2020-09-04 by Levi-Armstrong