tesseract-robotics/tesseract #198: Clang-tidy fails on 16.04

mpowelson, 2020-01-08:
I am getting cases where clang-tidy will fail when it shouldn't. Specifically I get unused parameter errors when they are actually used.

mpowelson, 2020-01-08:
According to @Levi-Armstrong this is a known issue with Clang-tidy. The debian for clang-tidy is missing the dependency on clang. The solution is to install clang-tidy 6 as well as clang-6.0.

gavanderhoorn, 2020-01-08:
Should (could) we explicitly add the dependency on whatever doesn't get installed?

If it ever gets fixed upstream, the additional dependency is already resolved. So it would not 'hurt'.

mpowelson, 2020-01-08:
I would think we probably could. I guess we would just add that to tesseract_macros.cmake? I suppose we could do a check for it the same way we do clang-tidy and look for versions that we know work.

gavanderhoorn, 2020-01-08:
I was thinking more of adding the separate dependencies (ie: clang and clang-tidy) to the package manifest of *some* package. That would allow `rosdep` to install them, if they're not present on the system already.

Are we right now depending on `clang-tidy` to pull in clang? If so, perhaps adding clang itself as a dependency would avoid the problem you report.

Levi-Armstrong, 2020-01-09:
These dependencies should probably be added to tesseract_common as build dependencies as @gavanderhoorn suggested.

Levi-Armstrong, 2020-02-15:
Addressed by #231

closed 2020-02-15 by Levi-Armstrong