tesseract-robotics/tesseract #205: Tesseract_python doesn't build cleanly on master

haudren, 2020-01-20:
I noticed that since the addition of `tesseract_python`, I have errors related to the fact that the default system SWIG is 3.0 while `tesseract_python` is 4.0

A few questions:
- I looked around and saw that:
https://github.com/Levi-Armstrong/tesseract_ext/commit/26d1dd62ae432afd1097af83d56ceb43b0c9c9cf
Adds SWIG 4.0 to the tree, which is nice, but the submodules are not up-to-date here. Moreover, even with this branch checked out for `tesseract_ext`, I had to run the build twice because `CMake` uses `find_program`  but when `swig` is done building, the path cache is not up-to-date, and the default `swig` still points to the system one... Not sure how to solve that last one.
- Moreover, I wonder how the Docker image for CI was built, it might be nice to have a Dockerfile here to make it easier to reproduce?
- Was there any specific reason why SWIG 4.0 was used? It seems like using 3.0 would be much easier to handle as it's part of default Ubuntu repositories. I personally prefer `pybind11` but that's another story altogether.

Levi-Armstrong, 2020-01-20:
Once tesseract_ext PR gets merged I will create a PR against tesseract updating the sub-modules. I typically put tesseract_ext in an extended workspace so I have not experienced this. I will clone down the repository with sub-modules and see if I experience the same issue.

Docker Image build:
```
FROM ros:melodic-perception
MAINTAINER Levi Armstrong

ARG DEBIAN_FRONTEND=noninteractive

RUN apt-get update && apt-get install -y --no-install-recommends \
    ros-melodic-ompl \
    python-catkin-tools \
    ros-melodic-octomap-msgs \
    ros-melodic-octomap \
    libassimp-dev \
    libtinyxml-dev

RUN git clone --branch master https://github.com/bulletphysics/bullet3.git
RUN mkdir /bullet3_build && \
    cd /bullet3_build && \
    cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_CXX_FLAGS="-w -std=c++14" -DCMAKE_C_FLAGS="-w" -DBUILD_UNIT_TESTS=OFF -DBUILD_SHARED_LIBS=ON -DBULLET2_MULTITHREADING=OFF ../bullet3 && \
    make -j8 && \
    make install

RUN git clone --branch master https://github.com/danfis/libccd.git
RUN mkdir /libccd_build && \
    cd /libccd_build && \
    cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_CXX_FLAGS="-w -std=c++14" -DCMAKE_C_FLAGS="-w" -DENABLE_DOUBLE_PRECISION=ON ../libccd && \
    make -j8 && \
    make install

ENV PKG_CONFIG_PATH /opt/ros/melodic/lib/pkgconfig
ENV LD_LIBRARY_PATH /opt/ros/melodic/lib
RUN git clone --branch master https://github.com/flexible-collision-library/fcl.git
RUN mkdir /fcl_build && \
    cd /fcl_build && \
    cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_CXX_FLAGS="-w -std=c++14" -DBUILD_TESTING=OFF ../fcl && \
    make -j8 && \
    make install

RUN apt-get install --no-install-recommends -y tar wget

RUN wget http://prdownloads.sourceforge.net/swig/swig-4.0.1.tar.gz
RUN tar -xvf swig-4.0.1.tar.gz
RUN cd /swig-4.0.1 && \
    ./configure && \
    make -j8 && \
    make install

RUN git clone --branch release-1.8.1 https://github.com/google/googletest.git
RUN mkdir /gtest_build && \
    cd /gtest_build && \
    cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_CXX_FLAGS="-w -std=c++14" -DBUILD_GMOCK=OFF -DBUILD_GTEST=ON -DBUILD_SHARED_LIBS=ON ../googletest && \
    make -j8 && \
    make install

RUN rm -rf /bullet3 /libccd /fcl /bullet3_build /libccd_build /fcl_build /swig-4.0.1 /googletest /gtest_build
RUN rm swig-4.0.1.tar.gz

RUN apt-get install --no-install-recommends -y ros-melodic-rviz ros-melodic-octomap-ros clang clang-tidy libomp-dev
RUN ln -s /usr/local/bin/swig /usr/local/bin/swig4.0
RUN ln -s /usr/local/bin/ccache-swig /usr/local/bin/ccache-swig4.0

RUN apt-get clean && \
    rm -rf /var/lib/apt/lists/*

CMD ["bash"]
```

> Was there any specific reason why SWIG 4.0 was used? 

Maybe @johnwason can provide details on why swig 4.0 was required.

johnwason, 2020-01-20:
SWIG 4.0 is needed because the older versions don't support C++11 or shared_ptr very well.

haudren, 2020-01-21:
Thanks for the answers!

@Levi-Armstrong : Maybe this Dockerfile should live in the repo somewhere? You can use a build-arg to switch between melodic / kinetic I think. I can try to help reproduce that bug later if you need.

@johnwason : Thanks for the reply, those features are essential so 3.0 is a no-go. Let's stick with 4.0 but it hurts to have to compile it from scratch :cry:

Levi-Armstrong, 2020-01-21:
@haudren I just created a [PR](https://github.com/ros-industrial/docker/pull/30) to add tesseract docker files to the ros-industrial docker repository.

haudren, 2020-01-22:
Thanks @Levi-Armstrong ! I tested with latest `tesseract_ext`, seems to build cleanly, so I guess we only need to update the submodule here and `tesseract_python` will build cleanly!

Levi-Armstrong, 2020-12-12:
Reopen if still an issue, but I believe it has been solved in master.

closed 2020-12-12 by Levi-Armstrong