tesseract-robotics/tesseract #206: Add support for SRDF model in tesseract_python

haudren, 2020-01-20:
I was playing around with `tesseract_python`, and noticed that one feature is missing (and apparently the code is commented in `tesseract.i` ): support for SRDF model.

I use this area of the code mostly to have access to the group states, that are quite useful when setting up planning problems in a ROS environment (typically set the initial configuration to a known pose).

I will try to get a corresponding PR in in the coming days, but this issue can serve for discussion before that.

Levi-Armstrong, 2020-01-20:
Yea, I believe @johnwason said there is issue with the library that was causing an issue with swig. I think it was because it had nested structs within the class. Feel free to restructure the srdf library.

Levi-Armstrong, 2020-02-12:
@haudren Has this been addressed?

haudren, 2020-02-14:
Yes, this was fixed by #208 .

closed 2020-02-14 by haudren