tesseract-robotics/tesseract #207: Move Tesseract_ros to its own package

mpowelson, 2020-01-20:
I know this has been discussed, and I believe there are some roadblocks to this. What are they and how do we plan on resolving them?

schornakj, 2020-01-23:
I think that the packages are fully separated now. When I build Tesseract for ROS2 I just put a `COLCON_IGNORE` file in `tesseract/tesseract_ros` and Colcon uses `tesseract_ros2` instead, which indicates that there aren't any dependencies between `tesseract_ros` and the rest of the core Tesseract packages.

Levi-Armstrong, 2020-01-30:
I came across this repository showing an example having ros1 and ros2 packages in the same repository. I created a [PR](https://github.com/romainreignier/share_ros1_ros2_lib_demo/pull/1) showing how to merge them into a single package so the package name does not change between versions of ros. I wanted to get your thoughts on using this approach to merge in tesseract_ros2 into this repository? @schornakj @mpowelson @marip8 @jrgnicho @jdlangs

jrgnicho, 2020-01-30:
@Levi-Armstrong  you proposed there looks fine to me, although just adding a COLCON_IGNORE file would be much simpler and it would still build under ROS1 as long as tesseract_ros2 remains in its own repo

gavanderhoorn, 2020-01-30:
Not using the ugly conditionals in package manifests would also be much less .. well .. *ugly* ;)

Levi-Armstrong, 2020-01-31:
Currently there are three options.

- Option 1:
  - Core Libraries in its own repository
  - ROS1 Libraries in its own repository
  - ROS2 Libraries in its own repository
- Option 2:
  - All in the same repository but ROS1 and ROS2 packages must have different names. Example tesseract_monitoring (ROS1) then tesseract_monitoring2 (ROS2). Also the CMakeLists.txt must check for Catkin and if not found do not build shown in the [original example](https://github.com/romainreignier/share_ros1_ros2_lib_demo).
- Option 3:
  - All in the same repository where ROS packages can have the same names between ros version. But it requires a conditional package.xml and different folder structure shown the [modified example](https://github.com/romainreignier/share_ros1_ros2_lib_demo/pull/1).

I would like to get everyone's opinion about which one they prefer. @gavanderhoorn  @schornakj @mpowelson @marip8 @jrgnicho @jdlangs

mpowelson, 2020-01-31:
I think my preference would be first Option 1. Then Option 3.

Option 1 allows non ROS users to simply not clone the unused repositories rather than having them there but not building them.

Option 3 is easy and I suppose is a bit more convenient. However, I'm already imagining either myself or code following in an IDE getting confused and changing the wrong headers. Plus, I would think in a year or two a large percentage of our projects will be in ROS2 making this structure more inconvenient than useful. Though I suppose we could just restructure to remove ROS1 at that point.

I'm not sure I have any reasons for not liking 2. It just seems messy to add 2 to the package name.

Do any of these options make a future release easier/harder?

gavanderhoorn, 2020-01-31:
I posted a (very) long comment this (my) morning, but for some reason it seems to have disappeared. :(

not sure what happened here.

mpowelson, 2020-01-31:
@gavanderhoorn I didn't see an email on my end.

jdlangs, 2020-01-31:
I agree that option 1 seems to be the cleanest but as a developer I am pretty scared of having to coordinate changes that would affect both the core libraries and the wrappers across multiple repositories. To me, the core and wrappers are inherently a single project, and so it makes sense for them to live in a single repo where they will always be in sync and versioned together.

So I actually prefer option 2 here. I also don't like putting the number 2 on package names though; my suggestion would be to use names like `tesseract_monitoring` and `tesseract_monitoring_ros1`, which also is a not-so-subtle hint that ROS-I has moved on from ROS1.

gavanderhoorn, 2020-01-31:
> To me, the core and wrappers are inherently a single project, and so it makes sense for them to live in a single repo where they will always be in sync and versioned together.

One argument against this is that we separate "core libraries" and wrappers exactly because we don't see them as "a single project", but a library with business logic and some other entity that makes that available in the context of a ROS node graph (ie: a wrapper node).

jdlangs, 2020-01-31:
Business logic changes are one thing but wouldn't any change to the client interface / API of the core require a corresponding change in the wrapper? Are they not tightly coupled at that layer?

gavanderhoorn, 2020-01-31:
Not necessarily. I would say that depends rather heavily of the type of change(s).

We could re-implement the entirety of a class without changing any of the interfaces, which would mean no changes in the wrapper either.

jdlangs, 2020-01-31:
Definitely the one-way dependency is important. My point is that if the dependency is such that changes to the interface of the core *require* changes to the wrapper, it strongly suggests all those changes should be happening in a unified way. Admittedly, I haven't dug much into a project that uses a structure like this so I don't have much intuition about how often this is the case.

Levi-Armstrong, 2020-02-07:
@jdlangs and @gavanderhoorn Thank you for the discussion. I personally have struggled with the two option discussed. Keeping everything in one repository is nice from a maintains perspective, but from a developers perspective it is nice to keep them isolated. I am still not sure what is the best way forward is. I think we will keep down the original path of isolating them in separate repositories and see how things go.

mpowelson, 2020-03-13:
As of #241 this repo doesn't depend on ROS. We now have essentially 4 repos
1) [Tesseract](https://github.com/ros-industrial-consortium/tesseract) - Core Cmake Libraries and Python wrappers
2) [Tesseract_ext](https://github.com/ros-industrial-consortium/tesseract_ext) - External packages
3) [Tesseract_ros](https://github.com/ros-industrial-consortium/tesseract_ros) - ROS 1 wrappers and interfaces with Python wrappers
4) [Tesseract_ros2](https://github.com/ros-industrial-consortium/tesseract_ros2) - ROS 2 wrappers and interfaces

closed 2020-03-13 by mpowelson