tesseract-robotics/tesseract #216: Undefined reference on Kinetic with Xenial

johnwason, 2020-01-27:
I am getting the following error when building on Kinetic with Xenial using `catkin_make_isolated`:

```
[ 72%] Linking CXX executable create_convex_hull
CMakeFiles/create_convex_hull.dir/src/create_convex_hull.cpp.o: In function `btConvexHullComputer::compute(double const*, int, int, double, double)':
create_convex_hull.cpp:(.text._ZN20btConvexHullComputer7computeEPKdiidd[_ZN20btConvexHullComputer7computeEPKdiidd]+0x52): undefined reference to `btConvexHullComputer::compute(void const*, bool, int, int, double, double)'
collect2: error: ld returned 1 exit status
CMakeFiles/create_convex_hull.dir/build.make:107: recipe for target 'create_convex_hull' failed
make[2]: *** [create_convex_hull] Error 1
CMakeFiles/Makefile2:141: recipe for target 'CMakeFiles/create_convex_hull.dir/all' failed
make[1]: *** [CMakeFiles/create_convex_hull.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

2scholz, 2020-01-27:
I ran into the same problem and was able to fix it by checking out the master branch of `tesseract_ext`.
I think the error was caused by #213.

mpowelson, 2020-01-27:
It looks like this is a problem in CI as well. [TrajOpt](https://travis-ci.com/ros-industrial-consortium/trajopt_ros/jobs/279981963) and [Tesseract](https://travis-ci.com/ros-industrial-consortium/tesseract/jobs/279535286) jobs are failing. @schornakj Any insight?

Levi-Armstrong, 2020-01-27:
We found that when trying to find contact distances between objects around 2mm bullet fails to find a contact. This is a result of using floats instead of doubles. We have updated tesseract external to build bullet with doubles and updated tesseract collision to use the correct build flags so it knows bullet is built with doubles. I will need to create a PR against bullet to improves its CMake files because it does not export this build flag so we have to add it to tesseract_collision for now. This is causing things to break.

Levi-Armstrong, 2020-01-28:
Also it looks like the submodule needed to be updated #217

closed 2020-01-30 by Levi-Armstrong