tesseract-robotics/tesseract #236: Add SharedPtr alias

mpowelson, 2020-02-25:
It seems that the ROS 2 Style is to use ::SharedPtr instead of ::Ptr. Is there a reason not to add those to our classes? We could also deprecate the other way like it seems ROS 2 has done, but that might be a bit ahead of ourselves. Something like this

```
  using Ptr [[deprecated]] = std::shared_ptr<SceneGraph>;
  using ConstPtr [[deprecated]] = std::shared_ptr<const SceneGraph>;
  using SharedPtr = std::shared_ptr<SceneGraph>;
  using ConstSharedPtr = std::shared_ptr<const SceneGraph>;

```

marip8, 2020-03-03:
I personally don't see much value in changing simply to match the ROS2 convention. The extra six characters seem cumbersome and don't really tell me anything I didn't already know about the pointer type

mpowelson, 2020-03-03:
I think I agree. I'd probably vote that we add both. It makes it nice that you can then find and replace ::Ptr with SharedPtr to port ROS 2 code. However I don't really feel strongly about it.

It does seem odd that this changed though. There must have been rationale. Does the Google style guide dictate it this way?

Levi-Armstrong, 2020-11-25:
I could not find anything in the google style guide, but found the ROS 2 documentation below. I don't care to much about this but I had to choose it would be the shorter version, though we could add both, but only use one internally.

[Pointer types](https://design.ros2.org/articles/generated_interfaces_cpp.html)

The struct contains typedefs for the four common pointer types plain pointer, std::shared_ptr, std::unique_ptr, std::weak_ptr. For each pointer type there a non-const and a const typedef:

```
    RawPtr and ConstRawPtr
    SharedPtr and ConstSharedPtr
    UniquePtr and ConstUniquePtr
    WeakPtr and ConstWeakPtr
```

Tessearct Equivalent

```
    RPtr and ConstRPtr
    Ptr and ConstPtr
    UPtr and ConstUPtr
    WPtr and ConstWPtr
```

closed 2020-12-12 by Levi-Armstrong