tesseract-robotics/tesseract #246: CMake Error finding opw_kinematics

bgd29, 2020-03-09:
Trying to build tesseract using catkin build, but I get errors because my workspace doesn't have tesseract dependencies (opw_kinematics, descartes_light etc) e.g. "Could not find a package configuration file provided by "opw_kinematics". If I manually add the dependecies I am able to build but the tesseract_ros_examples crash "error while loading shared libraries: libtrajopt_utils.so" .
I did clone the repo using --recursive, so I'm not sure what the correct workflow is or if there is something (obvious) I'm missing.

Levi-Armstrong, 2020-03-09:
You should be able to run the rosinstall file in the main directory which should pull in all of these dependencies.

Levi-Armstrong, 2020-03-19:
Did running the rosinstall resolve the issue?

bgd29, 2020-03-19:
Yes, rosinstall solved the problem I had above with the missing packages. I noticed tesseract_ros was removed from the repo, is that being reworked?

mpowelson, 2020-03-19:
It has been moved to https://github.com/ros-industrial-consortium/tesseract_ros and https://github.com/ros-industrial-consortium/tesseract_ros2 in order to better support ROS1/ROS2 compatibility

bgd29, 2020-03-20:
I'm still having the crash (seg fault) problem with any of the examples from /tesseract_ros_examples. Should I close this issue and open a new one over at https://github.com/ros-industrial-consortium/tesseract_ros ?

Levi-Armstrong, 2020-03-21:
Yea that would be better.

closed 2020-03-21 by bgd29