tesseract-robotics/tesseract #250: CMake Error : pkg configuration "tesseract_rosutils" not found

sduminy, 2020-03-13:
Dear all,
I cloned recursively this repository by using `git clone --recursive https://github.com/ros-industrial-consortium/tesseract` but when I tried to build my workspace, I got the following error :
```
Errors     << frasier_motion:cmake /home/segolene/ros_workspace/logs/frasier_motion/build.cmake.000.log                                      
CMake Error at /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by
  "tesseract_rosutils" with any of the following names:

    tesseract_rosutilsConfig.cmake
    tesseract_rosutils-config.cmake

  Add the installation prefix of "tesseract_rosutils" to CMAKE_PREFIX_PATH or
  set "tesseract_rosutils_DIR" to a directory containing one of the above
  files.  If "tesseract_rosutils" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:11 (find_package)
```
I am not sure, how to overcome this. 
Could someone help me ?
Thanks

DavidMerzJr, 2020-03-13:
The ROS packages associated with Tesseract were recently moved out of this repository in #241.  They can now be found in their own [repository](https://github.com/ros-industrial-consortium/tesseract_ros).  If you had written code relying on those packages (which it appears you did), then you will need to also clone down that new repository.  Please let us know whether that fixes your issue.

sduminy, 2020-03-14:
After cloning the `tesseract_ros` package, the build worked well.
Thank you very much !
I will close this issue.
Have a good day.

closed 2020-03-14 by sduminy