tesseract-robotics/tesseract #258: undefined reference to `tesseract_motion_planners::TrajOptMotionPlanner::solve(tesseract_motion_planners::PlannerResponse&, bool)'

jianye0428, 2020-03-18:
hello,
i clone the package tesseract and tesseract_ros, and install the packages according to dependencies.rosinstall.

but when I run catkin build and get the following error:
`/home/jianye/unicar_project_demo2/devel/.private/tesseract_ros_examples/lib/libtesseract_ros_examples_pick_and_place_example.so: undefined reference to tesseract_motion_planners::TrajOptMotionPlanner::solve(tesseract_motion_planners::PlannerResponse&, bool)'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/jianye/unicar_project_demo2/devel/.private/tesseract_ros_examples/lib/tesseract_ros_examples/tesseract_ros_examples_pick_and_place_example_node] Error 1
make[1]: *** [CMakeFiles/tesseract_ros_examples_pick_and_place_example_node.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2
`

what should I do next to correct the error ?
thanks
Jian

mpowelson, 2020-03-18:
Try checking out Tesseract at 3adbfaec06118175984b4b7d369ea188ca92c8bb. I think this was broken in #247.

marip8, 2020-03-30:
I think this is an artifact of having moved `tesseract_ros` to its own repository. The function signature of the planner base class `solve` method was changed in #247, and I didn't update the downstream packages. It seems to be fixed now in `tesseract_ros`, judging by the fact that it passes CI and looking at [this line](https://github.com/ros-industrial-consortium/tesseract_ros/blob/943c9e83f842bc9a39a1b8357fa3ff95a6feb824/tesseract_ros/tesseract_examples/tesseract_ros_examples/src/pick_and_place_example.cpp#L305) and [this line](https://github.com/ros-industrial-consortium/tesseract_ros/blob/943c9e83f842bc9a39a1b8357fa3ff95a6feb824/tesseract_ros/tesseract_examples/tesseract_ros_examples/src/pick_and_place_example.cpp#L510) where the `solve` function is called

jianye0428, 2020-04-03:
ihave installed the package again and it seems this problem has been fixed! thx!

closed 2020-04-03 by jianye0428