tesseract-robotics/tesseract #263: How to retrieve objects from the environment ?

sduminy, 2020-03-26:
Hi all,
I have been working on a pick & place task with object recognition. I am working from a code initially based on OpenRave and trying to switch to another library based on tesseract for motion planning. 
I would like to know if there is a method to retrieve object's position and size from the environment with the object's name (those features have been added previously during an objects detection phase.
I am looking into tessereact_environment source code but without success...
Does someone know if such a method is implemented on tesseract ?
Thanks,

mpowelson, 2020-03-26:
Perhaps [getLink](https://github.com/ros-industrial-consortium/tesseract/blob/a564b4d0d29d8783ae165d1e0d14fa7b94fe6b73/tesseract/tesseract_environment/include/tesseract_environment/core/environment.h#L177) is what you are looking for? Then the  link will have visual and collision elements associated with them which have some geometry type (e.g. box, cylinder, etc) .

sduminy, 2020-03-26:
Yes I am trying to work with that ! Do you know what method can I use with a tesseract_scene_graph::Link to get the pose and size of the object attached to this link ?

Levi-Armstrong, 2020-03-26:
I would us the environment getCurrentState or if you may also use the getState where you provide joint values. This will provide an object that contains a map of link names to transforms in world coordinate system. Then using getLink you have the geometry and the geometry pose relative to the links origin so you can calculate the location of the geometry in world coordinate system.

The size of the object will require you to calculate. For primitive shapes this is straight forward but for meshes and convex hulls you would need to do the calculation. Are you just wanting the bounding box?

mpowelson, 2020-03-26:
@Levi-Armstrong We could use some doxygen on EnvState https://github.com/ros-industrial-consortium/tesseract/blob/a564b4d0d29d8783ae165d1e0d14fa7b94fe6b73/tesseract/tesseract_environment/include/tesseract_environment/core/types.h#L54

sduminy, 2020-03-26:
Yes so actually the objects I want to retrieve the pose and size from, have first been added as link with the `tesseract_environment::addLink(tesseract_scene_graph::Link link, tesseract_scene_graph::Joint joint)` method.
Using `tesseract_environment::getLink(const std::string& name)` I should have the right link. But then, how can I extract the pose and size of it ?

mpowelson, 2020-03-26:
So if you want the link itself and all the geometry information associated with it (like size), you will need to do getLink. Presumably the size information you want is in either the visual or collision geometry.

If you want to find out where that link is based on the current state of the environment, you will need to call getCurrentState. Then it sounds like the transform map @Levi-Armstrong  mentioned and I linked above is in world frame.

sduminy, 2020-03-26:
Okay I will search into visual and collision then ! Thanks for all the information !

sduminy, 2020-03-27:
I would like to access data in an object of type `tesseract_geometry::Geometry`, which is the result of the method `tesseract_collision::CollisionShapesConst& getCollisionObjectGeometries(std::string obj_name)`. More specifically, I want to access to the x,y,z element of the object size.
Do you know what is the method ?

Levi-Armstrong, 2020-03-27:
It returns the base class type, so you will need to use the method getType() and based on the type of the object will need to cast it this type. Once you do that you will have access to all of the type size information.

sduminy, 2020-03-27:
Thank you a lot it worked !
Moving to the obj_pose which should be returned by the method `tesseract_collision::ContinuousContactManager::getCollisionObjectGeometriesTransforms` which returns an object of type `tesseract_common::VectorIsometry3d`, I have the same question. 
How could I retrieve the x,y,z components of the object's position ?

Levi-Armstrong, 2020-03-27:
So the vector of Transforms and vector of geometries are a one-to-one correlation. If you are interested in the first geometries transform you would do the following.

```
tesseract_common::VectorIsometry3d tf = a.getCollisionObjectGeometriesTransforms();
double x = tf[0].translation().x();
double y = tf[0].translation().y();
double z = tf[0].translation().z();
```

Note: tesseract_common::VectorIsometry3d is a vector of Eigen::Isometry3d matrices.

sduminy, 2020-03-27:
Okay thanks a lot !
This issue is solved, I will close it.

closed 2020-03-27 by sduminy