tesseract-robotics/tesseract #276: Tesseract Collision assert triggering

mpowelson, 2020-04-06:
I am getting an assert triggered [HERE](https://github.com/ros-industrial-consortium/tesseract/blob/02cd020bd73ebcb9fd356494d3874f07aedc1fa7/tesseract/tesseract_collision/include/tesseract_collision/core/common.h#L128) during a cylinder-cylinder check using the BulletDiscreteSimpleManager. Are the if statements set up correctly here?

Levi-Armstrong, 2020-04-06:
Yea, have you rebased on the latest? I ran into this when updating the unit tests for tesseract_collision in PR #274 and I believe I fixed it.

mpowelson, 2020-04-06:
I still seem to be getting it. Is there a reason why if(found) the contact test type should not be first?

Levi-Armstrong, 2020-04-06:
Which contact manager is causing the assert?

mpowelson, 2020-04-06:
BulletDiscreteSimpleManager. It happens when I run the benchmarks in debug.

Levi-Armstrong, 2020-04-06:
Does the benchmark run multiple threads?

mpowelson, 2020-04-06:
I'm not sure. I think it can.

mpowelson, 2020-04-06:
This was my own fault. I was not clearing the contact results before calling contactTest. Apparently this is not necessarily a problem (you may not notice in release), but will result in poor performance.

I had to change
```
  ContactResultMap result;
  for (auto _ : state)
  {
    info.contact_manager_->contactTest(result, info.contact_test_type_);
  }
```
to
```
  ContactResultMap result;
  for (auto _ : state)
  {
    result.clear();
    info.contact_manager_->contactTest(result, info.contact_test_type_);
  }
```

closed 2020-04-06 by mpowelson