tesseract-robotics/tesseract #287: Integrate toppra for time parameterization.

Levi-Armstrong, 2020-05-07:
https://github.com/hungpham2511/toppra

mpowelson, 2020-05-07:
@marrts  might have some useful code around somewhere. This was discussed in [this crs pull request](https://github.com/swri-robotics/collaborative-robotic-sanding/pull/36).

gavanderhoorn, 2020-05-07:
> Edit, I think ECOS is only in the python code. The new C++ API looks to use qpOASES

Exactly. I was just going to post a comment about this.

And it's all configurable as well.

marrts, 2020-05-07:
> @marrts might have some useful code around somewhere. This was discussed in [this crs pull request](https://github.com/swri-robotics/collaborative-robotic-sanding/pull/36).

I didn't look too much into toppra at that time and I ended up using [this modified code from moveit's iterative spline parameterization](https://github.com/schornakj/iterative_spline_parameterization) from @schornakj which has served well for me but doesn't control for tool speed at all, just joint velocity and acceleration.

Levi-Armstrong, 2020-05-07:
If they are integrating it using qpOASES could we just introduce it as a cost/constraint to our trajopt problem with the new IFOPT integration work? This way it may influence the final trajectory instead of a post trajectory operation.

mpowelson, 2020-09-28:
I think I know the answer to this, but has anyone by chance looked into this? TOPPRA is getting some votes of confidence over on https://github.com/ros-planning/moveit/issues/2183.

Levi-Armstrong, 2020-09-29:
I had looked at it a few months back, but have not testing anything.

closed 2020-12-30 by Levi-Armstrong