tesseract-robotics/tesseract #293: Planning with a subset of joints

hzyjerry, 2020-05-24:
Hi there,

I'm wondering if there is a way to only run tesseract planning with a subset of joints? I tried 

- specifying `passive_joint` in srdf, and looks like tesseract is able to pick it up, but tesseract planning still moves those joints
- setting the joints to `fixed` in urdf, which works, but feels a bit hacky.

Thanks!

hzyjerry, 2020-05-24:
Oops just found out that you can name a `group` of joints `manipulator` and only work with that using tesseract planning. Very neat!

closed 2020-05-24 by hzyjerry

Levi-Armstrong, 2020-05-25:
Glad you found a solution. When using a list of joints it will be represented as a kinematic tree. It also provides a method for updating all other joints in the environment in the kinematics object since the kinematic objects are not aware of other joints.