tesseract-robotics/tesseract #296: Improve management of external dependencies

schornakj, 2020-05-28:
This is a continuation of a discussion that started in [PR #20 in tesseract_ros2.](https://github.com/ros-industrial-consortium/tesseract_ros2/pull/20).

To briefly summarize:

- Currently `tesseract_ext` uses CMake's `AddExternalProject` function to clone and build Bullet, FCL, and some other packages. There might be a better way to approach this.
  - Benefits
    - Allows passing CMake flags to projects
    - Handles packages that do not contain a `package.xml` file
  - Drawbacks
    - Source code gets put in the build directory, which gets deleted if the workspace is cleaned
    - Intermingles the conceptually-distinct processes of _setting up_ and _building_ dependencies
  - An alternative via @gavanderhoorn:
  > Colcon actually supports specifying build flags for packages by including a [`.mixin`](https://colcon.readthedocs.io/en/released/reference/mixin-arguments.html) with the appropriate keys and values. Those can be stored in the repository and then the responsibilities are appropriately separated (`vcstool` fetches the sources, `colcon` builds them, flags are set by reading the `.mixin`s).
- Dependencies listed in `dependencies.rosinstall` are not installed by `wstool` when installing packages which include Tesseract in their own `.rosinstall` file, so if Tesseract gains new dependencies then projects that use Tesseract need to add these dependencies on their own.
  - (This might just be me misunderstanding `wstool` and `vcstool`, but I haven't had much success recursively evaluating source-repo dependencies).

gavanderhoorn, 2020-05-29:
There is no support for recursively retrieving sources using `wstool` or `vcstool`. See [](https://answers.ros.org/question/314789) for a ROS Answers Q&A about that topic (@VictorLamoine commented with a link to a bash script he wrote to work-around that).

I believe the idea is to use `rosinstall_generator` for those cases. That will generate a `.rosinstall` file for you including all the required dependencies to be built from source.

That does sort of require a `distribution.yaml` be created for the involved projects, containing the required information for `rosinstall_generator` to consume and process, so that may be a bit too much overhead (note: they don't need to be released, `rosinstall_generator` can work with source repositories as well -- in fact, this is what Clearpath used to do (I believe) with their "omnibus builds").

If Colcon is not possible/available, two other approaches:

 1. fork upstream, add a `package.xml`, update whatever flags you need in the `CMakeLists.txt` and build the fork as part of a ROS workspace
 1. create a "vendor package" (ie: `bullet3_vendor`), which essentially does what the `trajopt_ext` does now, but one vendor pkg builds one dependency. This would still use `AddExternalProject(..)` though

In both cases, the location of the repositories hosting the dependencies would be added to the `.rosinstall` files.

I've used both approaches, the first one works pretty well, and avoids the download step when cleaning and rebuilding a workspace.

The second approach is nice (from a minimal-package perspective), but does not solve the download problem.

Levi-Armstrong, 2020-06-12:
Doesn't an external workspace solve these issue? I typically have two workspace for every project and external workspace where I put packages I normally do not touch and then the main workspace with the packages I am working on.

> Dependencies listed in dependencies.rosinstall are not installed by wstool when installing packages which include Tesseract in their own .rosinstall file, so if Tesseract gains new dependencies then projects that use Tesseract need to add these dependencies on their own.

[Here](https://github.com/ros-planning/moveit_ci/blob/556b9f19db6346812d2a2027b411a3334c1b778c/travis.sh#L217-L236) is how MoveIt CI does this. It basically loops over every package pulled from the primary rosinstall and then runs wstool merge on all the rosinstalls and to create a single rosinstall.

Though it looks like some are starting to use [vcstool](https://github.com/dirk-thomas/vcstool) instead of wstool which my have the functionality we need.

gavanderhoorn, 2020-06-12:
> Doesn't an external workspace solve these issue? I typically have two workspace for every project and external workspace where I put packages I normally do not touch and then the main workspace with the packages I am working on.

The official name for your "external workspace" is an *underlay*.

And yes, that's what you would use an underlay for.

> [Here](https://github.com/ros-planning/moveit_ci/blob/556b9f19db6346812d2a2027b411a3334c1b778c/travis.sh#L217-L236) is how MoveIt CI does this. It basically loops over every package pulled from the [...]

This is a custom bit of script (similar to the one I linked to on ROS Answers).

Of course you could do something like this. The disadvantage of this approach however is it doesn't take care of any conflicts (ie: my dependency depends on something I also depend on, and now I have to source checkouts of the same thing ..).

closed 2020-12-11 by Levi-Armstrong