tesseract-robotics/tesseract #304: Planning in dynamic environment

hzyjerry, 2020-06-15:
Hi! I'm wondering if the current tesseract framework supports planning in dynamic environment? For instance, you want to attach a part to a vehicle that is moving fix-speed on a conveyer belt. It'd be a super nice feature that fits my use case.

If it's not supported yet, I wonder what changes in tesseract do you foresee that this function would entail?

Thank you for your insight!

Levi-Armstrong, 2020-06-16:
So Tesseract at the core was designed for this, so components like tesseract_collision, tesseract_environment, etc. fully support this dynamic functionality. The current motion planners function by taking a snap shot of the environment in time and create a robot trajectory. Then when executing the trajectory we assume the environment has not changes. On our system we currently leverage the contact monitory node in tesseract_ros to actively detect potential collisions and stop the systems.

We recently went through and updated core components of Tesseract and TrajOpt to potential achieve realtime motion planning shown [here](https://github.com/ros-industrial-consortium/tesseract_ros/pull/19). We are currently working on integrate trajopt_ifopt to simplify the use of this capability, but you could take this along with ros_control trajectory splicing functionality where you constantly tracking the object and updating the trajectory. If you know the speed of the object then you could predict its location and continuously update the trajectory until it is picked. 

> If it's not supported yet, I wonder what changes in tesseract do you foresee that this function would entail?

I believe most of the work would be around creating a controller to leverage the new trajopt_ifopt. We planned to start with the ros_control trajectory splicing with an ultimate goal of integrating the [control_toolbox](https://github.com/ethz-adrl/control-toolbox). If your interested in integrating this into Tesseract I would be happy to assist.

hzyjerry, 2020-06-16:
The realtime motion planning example is really cool! I'll definitely be playing with the example and reading up on the latest updates in the next few days.

The trajectory splicing method seems the most reasonable way to go. Earlier I was thinking of another perhaps more "hacky" way of dynamic trajOpt by:

1. making `n_steps` copies of tesseract environments, each set to i-th future state
2. running `SingleTimestepCollisionEvaluator` on each instance

this way you only run optimization once, but with predicted future states. I think the biggest drawback is it seems like a hack and the API would be awkward. Perhaps you could offer some comments on this approach.

mpowelson, 2020-06-16:
I will say that while we are pretty excited about the realtime planning as well, if you just need something to work in simulation you could probably get something working pretty easily using a variation of your approach. You could represent the conveyor as a prismatic in the urdf. Then you can place constraints on the conveyor's positions (or velocity) and plan to a frame attached to the conveyor link using TrajOpt DynamicCartesian terms. 

However, I am excited about the possibility of planning to moving targets in the manner Levi mentioned. I think it will open a lot of opportunities in all kinds of spaces. We'd be glad to get your help and/or see what your application looks like.

hzyjerry, 2020-06-16:
@mpowelson @Levi-Armstrong thank you for the awesome insights! Closing this issue as the original problem has been resolved.

closed 2020-06-16 by hzyjerry