tesseract-robotics/tesseract #317: Scaling issue from DAE files

hehonglu123, 2020-07-03:
URDF loader isn't correctly handling scaling from sketchup DAE files.
In Gazebo:
[image]
In Tesseract Viewer:
[image]
@johnwason

Levi-Armstrong, 2020-07-06:
Yea, it is possible that the scale is not being handled yet.

Levi-Armstrong, 2020-07-06:
@haudren Are you providing a scale in the urdf or are you suggesting that the dae file itself has a scale which is not being handled?

hehonglu123, 2020-07-06:
I tried the urdf with scale and without, not affecting the result (giant). I believe there's default scale in .dae files. .stl file works fine with the default scale; but with scale parameter in urdf, it's different than what shows up in gazebo.

Levi-Armstrong, 2020-07-06:
What visualizer are you using?

hehonglu123, 2020-07-06:
tesseract_viewer_python

johnwason, 2020-07-06:
I checked the output from the Mesh class, and the triangle positions are far larger than they should be. The tesseract_viewer_python passes the output directly from the Mesh class, and then adds the scale separately. I confirmed that the scale values passed to the viewer matched those specified in the URDF file. The issue seems to be a transform in the dae file that is not being applied to the position array when it is loaded.

Levi-Armstrong, 2020-07-07:
@johnwason are you working on a fix?

johnwason, 2020-07-07:
No, we were able to work around this problem by using STL files. The problem only seems to occur with sketchup exports.

Levi-Armstrong, 2020-07-07:
Do you mind posting the sketup dae and the stl?

hehonglu123, 2020-07-07:
stl: https://github.com/hehonglu123/RR_Project/blob/master/simulation/models/box2/box2.stl
dae: https://github.com/hehonglu123/RR_Project/blob/master/simulation/models/box2/model.dae

Levi-Armstrong, 2020-07-17:
I have found the issue and not sure how best to fix it.

https://github.com/ros-industrial-consortium/tesseract/blob/b79ea6bcad011bd17761d96cf50e2f0f1c5d35c3/tesseract/tesseract_geometry/include/tesseract_geometry/mesh_parser.h#L191-L196

and 

https://github.com/ros-industrial-consortium/tesseract/blob/b79ea6bcad011bd17761d96cf50e2f0f1c5d35c3/tesseract/tesseract_geometry/include/tesseract_geometry/mesh_parser.h#L279-L284

johnwason, 2020-07-17:
Yeah, the ROS software doesn't handle collada format properly. I suggest we deprecate it in favor of glTF since  collada is so ambiguous.

Does RViz have the same problem?

Levi-Armstrong, 2020-07-17:
RViz does have the same problem.

Levi-Armstrong, 2020-07-19:
I appears to be a assimp importer issues shown [here](https://github.com/assimp/assimp/issues/849).

johnwason, 2020-07-19:
I tracked down the rotation problem to the way RViz imports dae files:

https://github.com/ros-visualization/rviz/blob/291df89957e2f643a3fde28e5cb239fe1f6f5b73/src/rviz/mesh_loader.cpp#L240

I suspect that this is where the scaling issue is coming from as well. Unfortunately all ROS software has this issue so it is probably not possible to fix it now.

Levi-Armstrong, 2021-01-06:
@johnwason I believe the solution is to update the documentation that all models must be exported with Z-UP configuration to work correctly?

johnwason, 2021-01-06:
@Levi-Armstrong Yeah, that is probably the best solution. Blender can do some strange transforms during export to enforce the Y-up convention, which is the convention for game engines. Collada has a flag for Y-up or Z-up, but ROS handles Y-up inconsistently so it is best to ignore it. glTF2 is supposed to always be Y-up, but in practice most models are Z-up.

johnwason, 2021-01-06:
Another interesting tidbit, old style game engines used screen coordinates with 0,0 in the lower left corner, X to the right, Y up, and Z into the screen. This led to a left hand coordinate system. The inconsistent use of left and right handed coordinate systems make working with game engines annoying, although newer ones are using right handed coordinate systems.

simonschmeisser, 2021-12-05:
on the rviz side this might be fixed by https://github.com/ros-visualization/rviz/pull/1685

please leave a comment if your sample file works or not

rjoomen, 2023-07-25:
I think this can be closed.

burakaksoy, 2024-06-15:
Could not be sure whether to create a new issue or follow up from this one but I am reproducing the scaling issue. With scale parameter in URDF when specifying a mesh file (I tested with .obj and .dae files), the Tesseract somehow takes the **square** of the scaling parameter

Levi-Armstrong, 2024-06-15:
What version are you using?

burakaksoy, 2024-06-15:
Installed through pypi, 

Tesseract is v0.4.0.
The viewer is v0.2.5

Levi-Armstrong, 2024-06-18:
That is an old version of Tesseract so likely fixed in the latest version.

johnwason, 2024-06-18:
The version of Tesseract is 0.21.5 so I don't think anything has changed to affect this issue.

closed 2024-06-23 by Levi-Armstrong