tesseract-robotics/tesseract #333: Leverage Composite Start Waypoint for Motion Planner Results and Seed

Levi-Armstrong, 2020-07-15:
Currently Composite Instruction has a setStartWaypoint for setting the start state for the motion planners. 

Request:

- Composite
  - Start Waypoint
  - Plan Instruction
  - Plan Instruction
  - Plan Instruction

Seed/Results:

- Composite (Has not start waypoint captured in the move instrucitons)
  - Composite
    - Move Instruction (Corresponds to the start waypoint in the request)
    - Move Instruction
    - Move Instruction
  - Composite
    - Move Instruction
    - Move Instruction
  - Composite
    - Move Instruction
    - Move Instruction

Recommend changing this so they align 

- Composite (Has not start waypoint captured in the move instruction)
  - Start Waypoint (Most likely will always be a joint waypoint)
  - Composite
    - Move Instruction
    - Move Instruction
  - Composite
    - Move Instruction
    - Move Instruction
  - Composite
    - Move Instruction
    - Move Instruction

Levi-Armstrong, 2020-07-15:
@mpowelson Thoughts?

mpowelson, 2020-07-15:
This should be Start Instruction right? Such that it can be stored in a CompositeInstruction. If so, that is basically just a MoveInstruction right?

mpowelson, 2020-07-15:
Or are you just saying that we should fill out the start waypoint in the seed composite?

Levi-Armstrong, 2020-07-15:
> Or are you just saying that we should fill out the start waypoint in the seed composite?

Yea and remove the first move instruction in the first composite. I imagine eventually we will change the composite to have a setStartState instead of setStartWaypoint where the start state is a class which includes a waypoint but may also include other start information.

mpowelson, 2020-07-15:
What if we added a Placeholder instruction

Before planning
Instruction:
- Composite: 1
   - Composite: 1.1
       - Move Instruction (start state)
       - Plan Instruction
   - Composite: 1.2
       - Placeholder Instruction
       - Plan Instruction

Seed:
- Composite: 1
   - Composite: 1.1
       - empty
       - empty
   - Composite: 1.2
       - empty
       - empty

After Plan of 1.1
Instruction:
- Composite: 1
   - Composite: 1.1
       - Move Instruction (start state)
       - Plan Instruction
   - Composite: 1.2
       - Move Instruction
       - Plan Instruction

Seed:
- Composite: 1
   - Composite: 1.1
       - Move Instruction
       - Move Instruction
   - Composite: 1.2
       - empty
       - empty

After plan of 1.2
After Plan of 1.1
Instruction:
- Composite: 1
   - Composite: 1.1
       - Move Instruction (start state)
       - Plan Instruction
   - Composite: 1.2
       - Move Instruction
       - Plan Instruction

Seed:
- Composite: 1
   - Composite: 1.1
       - Move Instruction
       - Move Instruction
   - Composite: 1.2
       - Move Instruction
       - Move Instruction

closed 2020-07-29 by Levi-Armstrong