tesseract-robotics/tesseract #339: Feature Command Language TODO's

Levi-Armstrong, 2020-07-29:
- [x] Fix broken unit tests
- [ ] Implement the task validators for 
  - [ ] Format check (for checking that seed matches instruction)
  - [ ] Discrete Collision
  - [ ] Continous collision
  - [ ] etc
- [x] Implement the rest of the step generators
  - [x] Implement fixed size joint interpolate generators
  - [x] Implement fixed size assign generators
  - [ ] Implement longest valid segment generators
- [ ] Address required todos placed everywhere in the code.
  - [x] tesseract_motion_planners
  - [ ] tesseract_process_managers
- [ ] Unit tests. I tried to structure everything such that components are easily testable. However, not many of them are currently tested. 
  - [x] Add unit test for fixed size joint interpolate generators
  - [ ] Add unit test for fixed size assign generators
  - [ ] Add unit test for longest valid segment generators
  - [ ] Add unit test for ompl problem generator
  - [ ] Add unit test for trajopt problem generator
  - [ ] Add unit test for descartes problem generator

mpowelson, 2020-08-18:
@Levi-Armstrong One issue I have not dug into, using the default config OMPL says that "Start state is in collision" when it still has a position distance (one example is 0.016223). I imagine we need to change the defaults such that the precheck sets a threshold of 0.

mpowelson, 2020-08-18:
> @Levi-Armstrong One issue I have not dug into, using the default config OMPL says that "Start state is in collision" when it still has a position distance (one example is 0.016223). I imagine we need to change the defaults such that the precheck sets a threshold of 0.

It looks like the default distance threshold is set to 0.025. For OMPL shouldn't it usually be 0?

Levi-Armstrong, 2020-08-18:
Yea I think all of the planners default to 0.025. I am good with changing the defaults.

mpowelson, 2020-08-24:
Do you think time parameterization (and any other planning adapter like things) should be profile-aware? If so, do we want to go the whole way of wrapping it as a motion planner or just add it to the process generator? It's only got 2 parameters, but it seems like it might be nice to set them via profile. e.g. set the scaling factor really low on FINE moves but set to 1 on FREESPACE moves.

mpowelson, 2020-08-24:
Of course I think your only option then is to assume that the robot stops whenever the profile changes and run ISP on the sections separately.

mpowelson, 2020-08-25:
> Do you think time parameterization (and any other planning adapter like things) should be profile-aware? If so, do we want to go the whole way of wrapping it as a motion planner or just add it to [...]

After offline discussion we decided to address this in 2 ways. First, we will expose process generators in the taskflow generators allowing things like parameters to be changed after adding them. Second, we will profiles to ISP in the process generator but not fully wrap it as a motion planner since that seemed like overkill.

mpowelson, 2020-08-28:
@Levi-Armstrong How would you implement something like the MoveIt FixStartStateBounds? The issue is that we would be changing the ProcessInput instruction (which is const) rather than the seed, so I don't think it would fit into our current process pipeline. A couple of options I see

1. Push it to the caller. Just tell people not to send the pipeline requests that are invalid
2. Add it to the process managers. Sort of the same as 1. Just take the input, copy it, and fix any issues with it.
3. Make instruction non-const. 
4. Something else. We could perhaps make a specific set of processGenerators that are allowed to change the instructions using const_cast or something.

mpowelson, 2020-08-28:
Ah.. That might do it as long as its only the start and end that you are trying to fix.

mpowelson, 2020-08-28:
Do all of the planners respect that currently?

Levi-Armstrong, 2020-08-28:
What ever creates the planning request that gets sent to the planner does.

mpowelson, 2020-08-28:
Ok. I'll make a process generator that overrides the start or end and clamps it to the joint limits if they are outside.

mpowelson, 2020-09-04:
@Levi-Armstrong I added a visualization process to the wishlist. Perhaps this could be an observer or we could build it into our motion planning process generator rather than its own thing. However, it would be nice to be able to step through the results of each process somehow. 

If we can serialize the command language, it could also be an offline tool and each process could just dump the input and output to a file.

Levi-Armstrong, 2020-09-04:
I think we could create a visualization container that mirrors the results and every step through the taskflow it appends the results. What do you think?

- Componsite
  - Composite (to start)
  - Composite (Raster 1)
  - Composite (Transition)
  - Composite (Raster 2)
  - Composite (to end)

Visualization
- Componsite
  - Composite (to start)
    - Composite (Simple Planner Results)
    - Composite (OMPL Planner Resutls)
    - Composite (TrajOpt Planner Results)
    - Composite (Time Parameterization Results)
  - Composite (Raster 1)
    - Composite (Simple Planner Results)
    - Composite (OMPL Planner Resutls)
    - Composite (TrajOpt Planner Results)
    - Composite (Time Parameterization Results)
  - Composite (Transition)
    - Composite (Simple Planner Results)
    - Composite (Descartes Planner Resutls)
    - Composite (TrajOpt Planner Results)
    - Composite (Time Parameterization Results)
  - Composite (Raster 2)
    - Composite (Simple Planner Results)
    - Composite (Descartes Planner Resutls)
    - Composite (TrajOpt Planner Results)
    - Composite (Time Parameterization Results)
  - Composite (to end)
    - Composite (Simple Planner Results)
    - Composite (OMPL Planner Resutls)
    - Composite (TrajOpt Planner Results)
    - Composite (Time Parameterization Results)

mpowelson, 2020-09-04:
That seems like it might make it difficult to view all of the results for each step of the process. You couldn't just flatten them and then visualize like you could if you just stored a vector of the top level composite.

Also, I think I'd avoid putting this in "Core" code. I'm thinking we could write an observer that we setup with a const reference to the ProcessInput. Then it just watches when we enter and leave each process and copies the state of the ProcessInput into a datastructure at the right time. The only difficult piece there is we'd have to do something to make sure we have time to copy the results before the planners overwrite them. Perhaps we add a std::shared_mutex& to the ProcessInput? Observes are readers, processes are writers?

mpowelson, 2020-09-06:
@Levi-Armstrong I'm not sure when I'll get to finish this up, but I made a small start toward a trajopt_ifopt planner here. Feel free to take it or leave it if you decide to tackle it.

https://github.com/mpowelson/tesseract/tree/feature/trajopt_ifopt

Levi-Armstrong, 2020-09-06:
@mpowelson Thank you starting on this. I am currently working on serialization of the kinematics data stored in the tesseract to fully be able to construct a tesseract object. I Should have this wrapped up today or tomorrow and may get to the trajopt_ifopt late next week. I will reach out before I start.

mpowelson, 2020-09-10:
One thing I have run into is that debugging flow of the graph taskflow can be difficult, and errors occur at plan time rather than construction time (when nullptrs start getting referenced). Perhaps we should add a todo to check the input to the process manager at construction time and print some useful errors (rather than segfaulting)

mpowelson, 2020-09-10:
> I think we could create a visualization container that mirrors the results and every step through the taskflow it appends the results. What do you think?
> 
> * Componsite
>   
> ...

One other thing I think we discussed somewhere but I don't see on the list is logging error messages for each step. For instance, if the process manager says "planning failed", I want to be able to look back and see what each step said went wrong. This could be in the same datastructure, but I think error messages might be more core and should be a feature built into the process manager system.

Levi-Armstrong, 2020-09-10:
> One thing I have run into is that debugging flow of the graph taskflow can be difficult, and errors occur at plan time rather than construction time (when nullptrs start getting referenced). Perhaps [...]

I agree, I usually end up running it in a single thread and walking through the process.

Levi-Armstrong, 2020-09-11:
In the latest PR I updated the process managers so you at least know what ProcessInput failed, but need to figure out a better way to get everything back out.

mpowelson, 2020-09-11:
Yeah. I'm thinking I want to make TrajOpt smart enough to tell you why it failed for instance. Then the process managers would have an even higher level data structure. Semantically I'm wanting something like

- Interpolator: Succeeded
- Collision check: Failed - timestep: 0, Links: link_1, link_7 Dist: -0.2
- TrajOpt: Failed - Collision constraint violated. link_1 and link_7 are in collision at the fixed start state.

Levi-Armstrong, 2020-09-14:
Let's extend the status code class to support detailed description. I have not looked but I think that should allow you to flow the in formation up.

mpowelson, 2020-09-18:
Another feature request. We should take the plot function in problem_description in trajopt and make it a more general "plot collision results" function. Then we could add it to the contact check processes. I'm running into a case where trajopt passes with a collision constraint, but the contact check function fails.

mpowelson, 2020-09-18:
One other comment. The SimpleProcessManager must be constructed with a unique pointer to the taskflow generator. If we ever need a Tesseract inside the taskflow generator (eg to set up some visualization) then the Tesseract is usually not initialized since that has an init that has to be called whereas SimpleProcessManager has to go in the initializer list. I realize that ideally they would get those from the ProcessInput, but I don't know if that is always possible with current code.

Levi-Armstrong, 2020-09-18:
> Another feature request. We should take the plot function in problem_description in trajopt and make it a more general "plot collision results" function. Then we could add it to the contact check [...]

I just ran into this yesterday. The issue is that the contact have a default longest valid segment length of max integer so I have changed to be 0.1 and now my motion planning passes the post check. We will need to add a profile for this if there is not already one to make it configurable via motion request.

Levi-Armstrong, 2020-09-18:
> One other comment. The SimpleProcessManager must be constructed with a unique pointer to the taskflow generator. If we ever need a Tesseract inside the taskflow generator (eg to set up some [...]

If the tesseract object is always in the ProcessInput does that elevate the concern? I believe it should always be present

Slightly related, I have created a tesseract_monitor_interface on the ROS side that makes it easy for interacting with the Tesseract Monitor when your code does not have its own monitor. It can fully construct EnvState, Environment, Tesseract along with applyCommands to the Top level Monitor.

mpowelson, 2020-09-18:
I'm not sure I understand the tesseract monitor part. I need to rebase up to your latest changes. I'm a bit behind.

The issue I was running into is that in order to create a plotter for trajopt I needed the joint names which are hard to get at initialization since all of the other datastructures that parse that information haven't done so yet. This perhaps could be solved similarly to how we do in trajopt. The process manager could have a visualizer and each process could have an optional plot function.

mpowelson, 2020-09-18:
Though I don't really like making plotting that core to the system

mpowelson, 2020-09-18:
With the new kinematics manager can a ProcessInput now be fully serialized? If so, I am very interested in trying it out.

Levi-Armstrong, 2020-09-18:
> With the new kinematics manager can a ProcessInput now be fully serialized? If so, I am very interested in trying it out.

I believe so, If you take a look at the TesseractMonitorInterface you will find code where I fully construct a Tesseract Object from a message. Currently the only thing that is not passed is kinematics plugins like IKfast that you have loaded. This is the last remaining piece for full serialization through the Tesseract Framework. I plan to model the plugin loading similar to how Ignition handles it.

mpowelson, 2020-09-24:
> > With the new kinematics manager can a ProcessInput now be fully serialized? If so, I am very interested in trying it out.
> 
> I believe so, If you take a look at the TesseractMonitorInterface you will find code where I fully construct a Tesseract Object from a message. 

For those interested. See this PR https://github.com/ros-industrial-consortium/tesseract_ros/pull/41

mpowelson, 2020-10-20:
@Levi-Armstrong What would you think about starting a github project for getting feature/CommandLanguage merged in? I'm really not sure what is on our roadmap for getting that relatively stable. We could tag a minor release before merging and even keep a legacy branch around for a while if we need to. It seems that Tesseract/Trajopt has been getting some more traffic the last few weeks. It might be good to get people using the latest.

Levi-Armstrong, 2020-10-20:
Sound great!

mpowelson, 2020-10-20:
@Levi-Armstrong I created the project. Not sure what all we need to add.

Levi-Armstrong, 2020-10-20:
I start adding things and we can label things as required versus nice to have for a 1.0 release.

Levi-Armstrong, 2021-01-09:
remaining items have been moved to the Unit Test TODO #297

closed 2021-01-09 by Levi-Armstrong