tesseract-robotics/tesseract #343: Parsing box with no size errors

mpowelson, 2020-08-05:
I'm not necessarily recommending we change this, but it looks like urdfdom is happy to parse a box with dimensions of 0. Tesseract does not. It returns a fairly generic `ErrorAttributeSizeConversion` error. 

Perhaps we could give a more specific error message to aid in tracking down what exactly is the problem with "legacy" urdfs.

Levi-Armstrong, 2020-08-06:
Sure, just need to add a new error code and test.

mpowelson, 2020-08-06:
I'm not sure when I will get to this, but it should be minimal. The code that needs to be changed is here
https://github.com/ros-industrial-consortium/tesseract/blob/master/tesseract/tesseract_urdf/include/tesseract_urdf/box.h

Levi-Armstrong, 2022-01-26:
This has been addressed.

closed 2022-01-26 by Levi-Armstrong