tesseract-robotics/tesseract #347: Feature Request: Add Init Command to Environment Commands

mpowelson, 2020-08-14:
The Environment maintains a command history keeps track of changes applied and allows other copies of the environment to replicate them. This has the potential to become burdensome if a system is kept running for long periods of time with frequent changes (e.g. tool changes or adding collision objects to the environment). 

This feature would add an init command that tells the environment to save its current state into some serializable form and then restart the command history. The idea would be that on some regular interval, the main environment monitor would squash the history and send the command to all of the following environments to reinitialize using the current state.

Note, a start at this was made with the introduction of the TesseractInitInfo which essentially contains all of the ways to initialize a scene graph. However that was never fully integrated.

Levi-Armstrong, 2020-08-20:
@mpowelson With the addition of the SceneGraph Command is this resolved? Also do we need to retain the TesseractInitInfo anymore?

mpowelson, 2020-08-20:
I don't know about the init info. I suppose keeping that information can't hurt. Do things like manipulator groups get serialized? I suppose what I am really wanting is a way to keep a bunch of Tesseracts across ROS nodes in sync. If I initialized a Tesseract with the deserialized scene graph, I'm thinking some information in the URDF/SRDF is still lost.

mpowelson, 2020-08-20:
Also, the piece about squashing environment changes every x minutes or when the history gets to be y long still needs to be done I believe.

Levi-Armstrong, 2020-08-20:
> I don't know about the init info. I suppose keeping that information can't hurt. Do things like manipulator groups get serialized? I suppose what I am really wanting is a way to keep a bunch of [...]

Ah. yea need to serialize different components of the srdf. The allowed collision matrix are in the scene graph but need to add manipulator groups, tcp and opw to the Tesseract State message. I will start working on this.

Levi-Armstrong, 2020-09-01:
@mpowelson I am in the process of updating the Environment Monitor and need to figure out how best to serialize the kinematics with change history. Currently the only component that supports change history is the Environment which does contain item from the srdf (the allowed collision matrix), would it make sense to add the kinematics factories to the environment? The main reason for this, is then all changes can be tracked withing the Environment history. If not, then another data structure will need to be created that tracks the kinematic changes, which may be changing the Tesseract class to have methods for adding and removing kinematics objects and getters for the kinematics and it tracks the kinematics changes. Then the Tesseract state would publish two revision numbers one for the environment and one for the kinematics objects. Let me know what you think.

schornakj, 2020-09-01:
One issue we've run into with having a "local" Tesseract that monitors the main environment is that it currently relies on the revision ID always staying in sync for each environment update. If the ID gets out of sync (missed an update, one of the Tesseracts was reset, etc.) then it isn't possible to update the local environment anymore, with rather disruptive consequences for motion planning and collision checking.

I think that in situations where we're monitoring a main environment we want to just have the local environment match the state of the main environment rather than trying to exactly apply the series of updates and modifications that produced its current state. Something to keep in mind as we're implementing this.

Levi-Armstrong, 2020-09-01:
The current proposed switch would be to only leverage the TesseactState Topic for the revision number of the monitored environment. When creating the EnvironmentMonitor you provide a namespace for the environment and all topics/services are advertised /namespace/topic or /namespace/service. Then if you create an Environment monitor which does only monitors an environment (namespace), in the monitor environment callback it checks the revision and based on two cases it does the following. If the monitored environment is greater than its internal state it will perform a service call to the monitored environment get the changes and apply to its internal state. The second case is when it has a larger environment revision than the one it monitors and in this case it would call the service on the monitored environment to apply the changes. This way everything is keep in sync and with these changes it should not run into the case you mentioned. @schornakj What are your thoughts with this approach?

schornakj, 2020-09-01:
I think the behavior I'd prefer is that if the upstream environment ends up with a lower revision number (for example if the main EnvironmentMonitor node was shut down and restarted) then all the dependent environments should reset to match the newest state of the upstream environment, rather than pushing their latest state back to the upstream environment. It sounds like this is something we could expose as a setting when initializing the EnvironmentMonitors.

Levi-Armstrong, 2020-12-11:
Manipulator manager is now owned by the environment and changes are being stored in the command history.

closed 2020-12-11 by Levi-Armstrong