tesseract-robotics/tesseract #354: [feature/CommandLanguage] ROS2 class_loader API change breaks tesseract_visualization

schornakj, 2020-08-20:
The line below in `tesseract_visualization/src/visualization_loader.cpp` fails in ROS2:

https://github.com/ros-industrial-consortium/tesseract/blob/dc987d5bf64f230ad315c47e682f2bd79bf42f35/tesseract/tesseract_visualization/src/visualization_loader.cpp#L91

```
/home/joe/workspaces/colcon_ws/src/tesseract/tesseract/tesseract_visualization/src/visualization_loader.cpp: In member function 'tesseract_visualization::Visualization::Ptr tesseract_visualization::VisualizationLoader::get()':
/home/joe/workspaces/colcon_ws/src/tesseract/tesseract/tesseract_visualization/src/visualization_loader.cpp:91:25: error: 'using element_type = class class_loader::ClassLoader {aka class class_loader::ClassLoader}' has no member named 'createSharedInstance'; did you mean 'createRawInstance'?
         return loader_->createSharedInstance<tesseract_visualization::Visualization>(derived_class_);
                         ^~~~~~~~~~~~~~~~~~~~
                         createRawInstance
/home/joe/workspaces/colcon_ws/src/tesseract/tesseract/tesseract_visualization/src/visualization_loader.cpp:91:84: error: expected primary-expression before '>' token
         return loader_->createSharedInstance<tesseract_visualization::Visualization>(derived_class_);
                                                                                    ^
make[2]: *** [CMakeFiles/tesseract_visualization.dir/src/visualization_loader.cpp.o] Error 1
make[1]: *** [CMakeFiles/tesseract_visualization.dir/all] Error 2
make: *** [all] Error 2
```

I think this is because [this function was renamed to `createInstance`](https://github.com/ros/class_loader/blob/376005f173f34df8a2065df89881e719b4863833/include/class_loader/class_loader.hpp#L124-L131) in the ROS2 version of the library. Compare to [the equivalent ROS1 version of this function named `createSharedInstance`](https://github.com/ros/class_loader/blob/812744d284cd48cf575a15389d106f5d4c8903c0/include/class_loader/class_loader.hpp#L126-L132).

Levi-Armstrong, 2020-08-21:
I thought class loader was ros agnostic?

Levi-Armstrong, 2020-08-21:
I think we can switch to the Unique version. The only reason I did not initially use that was it was not available for kinetic. I think I should be able to check the version of class loader and switch for the kinetic.

schornakj, 2020-08-21:
> I thought class loader was ros agnostic?

I was under that impression as well. IMO it would have made more sense to keep the function signatures the same so that contemporary ROS1 and ROS2 releases could use the same API.

> I think we can switch to the Unique version. The only reason I did not initially use that was it was not available for kinetic. I think I should be able to check the version of class loader and switch for the kinetic.

Sounds reasonable!

Levi-Armstrong, 2020-09-04:
This should be fixed in the feature/CommandLanguage branch, but need to create a patch for master. It looks like for ros2 they have createInstance which returns a shared pointer opposed to createSharedInstance.

Levi-Armstrong, 2021-01-09:
Now that feature/CommandLanguage branch has been merged into master this should be resolved.

closed 2021-01-09 by Levi-Armstrong