tesseract-robotics/tesseract #368: Default Taskflow Discussion

Levi-Armstrong, 2020-08-28:
Currently we has a set of default taskflows.

1. TrajOpt Taskflow
    - This can either take a single instruction or a composite instruction containing child composites.
    - It will run (interpolator, trajopt, collision check, time parameterization) in series.
2. OMPL Taskflow
    - This can either take a single instruction or a composite instruction containing child composites.
    - It will run (interpolator, trajopt, collision check, time parameterization) in series.
3. Descartes Taskflow
    - This can either take a single instruction or a composite instruction containing child composites.
    - It will run (interpolator, trajopt, collision check, time parameterization) in series.
4. Simple Freespace Taskflow
     - This can either take a single instruction or a composite instruction containing child composites.
     - It will run (interpolator, ompl, trajopt, collision check, time parameterization) in series.
5. Simple Cartesian Taskflow
     - This can either take a single instruction or a composite instruction containing child composites.
     - It will run (interpolator, descartes, trajopt, collision check, time parameterization) in series.
6. Simple Raster Taskflow
     - This requires a composite instruction that follows a pattern
     - It will run (interpolator, descartes, trajopt, collision check, time parameterization) in series for rasters.
     - It will run (interpolator, ompl, trajopt, collision check, time parameterization) in series for transitions, from start, to end.

Currently the Simple Raster Taskflow creates dual transitions, but would like to discuss removing this from the Simple Raster Taskflow and creating another taskflow to handle this. @mpowelson what are your thoughts?

Additional taskflows proposed.

1. Simple Raster With Approach and Departures Taskflow
     - This requires a composite instruction that follows a pattern
     - This can either take a single instruction or a composite instruction containing child composites.
     - It will run (interpolator, descartes, trajopt, collision check, time parameterization) in series for rasters, approach and departures.
     - It will run (interpolator, ompl, trajopt, collision check, time parameterization) in series for transitions, from start, to end.
2. Global Raster Taskflow
     - This requires a composite instruction that follows a pattern
     - It will run (interpolator, descartes) on the whole trajectory first then the following
     - It will run (descartes, trajopt, collision check, time parameterization) in series for rasters.
     - It will run (ompl, trajopt, collision check, time parameterization) in series for transitions, from start, to end.
3. Global Raster With Approach and Departure Taskflow
     - This requires a composite instruction that follows a pattern
     - It will run (interpolator, descartes) on the whole trajectory first then the following
     - It will run (descartes, trajopt, collision check, time parameterization) in series for rasters, approach and departure.
     - It will run (ompl, trajopt, collision check, time parameterization) in series for transitions, from start, to end.

mpowelson, 2020-09-01:
One comment I have is that this is adding a lot of code to maintain. While I am all for open sourcing capability and making it as easy as possible to get something to work, I wonder if these would be better suited in an examples repo.

Levi-Armstrong, 2020-09-01:
That is true. The hope is in the end that these would just be represented in a config file which get dynamically generated. This way we only have to maintain the tool that can parse the xml and build the graph taskflow.

closed 2020-09-04 by Levi-Armstrong